EdgeTX/edgetx #2672: Owner Registration ID allows no "-" character in Companion radio settings

ParkerEde, 2022-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Currently it is not possible to enter a character such as a "-" in the radio settings under "Owner Registration ID". However, this is possible in the simulator as well as in the radio firmware.

### Expected Behavior

It should be possible to use the same characters in Companion that are allowed in the radio firmware.

### Steps To Reproduce

1. start Companion
2. create a new document
3. enter radio settings
4. try to enter abc-def

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild 8ce3612f

pfeerick, 2022-11-03:
Judging from OTX firmware, it looks like the punctuation is valid... ie. `-_,.` are allowable, plus maybe a few more.

elecpower, 2022-11-06:
Further investigation:
- b&w ie used X9D+ 2019 libsim and only valid characters are space, hyphen, 0-9, A-Z and a-z
- color ie used Radio Master TX16S libsim and appears to be no validation

So converting from color to b&w will also be an issue.

So it would appear the validation rules lies somewhere between these extremes. So need someone in the know or with factory firmware for Frsky b&w and ETHOS to advise. The ETHOS manual doesn't state the validation except max 8 chars.

pfeerick, 2022-11-07:
Oh, interesting... so I looked at both the X10E and X9D+2019 sims this time for OTX, and both allow for space, a-z (lower and upper), 0-9, underscore, hyphen, period, comma ... so my theory atm is that ETX B&W isn't allowing for all of the characters.. but will input.

Also had a look at the ETHOS simulator for X10E... and it allowed me to select from a much wider range of characters... so that doesn't really help either! :-O

https://user-images.githubusercontent.com/5500713/200207416-f8b7a248-159f-4d37-a594-6aaab6b14f27.mp4

elecpower, 2022-11-07:
The b&w limitation may have something to do with zchar which we have now done away with or I'm barking up the wrong tree again :-)
We do need some validation rules as we store the field in yaml so for example allowing quotes and hyphens without 'treatment' will likely cause other issues especially since we are using two different yaml writers and parsers.
I'll create a draft PR with a proposed explicit validator in the Companion code and remove the current validation from the ui design.
We can then ponder further and standardise across Companion and all radios.

PR #2687 (2022-12-10): fix(cpn): ACCESS owner registration id validation (#2672)
Fixes #2672

Summary of changes:
- add standard name validation
- hide label and field if radio does not support ACCESS

Note: decision required on inconsistent validation across radios
closed 2022-12-10 by pfeerick