EdgeTX/edgetx #2673: Using the trim buttons on Zorro results in a restart in ETX 2.7.1

msX99, 2022-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In ETX 2.7.0 I can use the trim buttons (for example for ailerons) on the Radiomaster Zorro without any issue.
In ETX 2.7.1 using any of the trim buttons crashes the software and results in an immediate restart. Trim is not applied.

This behavior can be verified with a new model (no special settings required).

### Expected Behavior

In ETX 2.7.1 the trim buttons should work without crashing the software.

### Steps To Reproduce

- Create a new model
- Return to the main screen
- Press any of the trim buttons on the Zorro TX

### Version

2.7.1

### Transmitter

Radiomaster Zorro

raphaelcoeffic, 2022-11-05:
@msX99 there must be something *special* about possibly your radio settings, as I cannot reproduce it here.

msX99, 2022-11-05:
I have renamed MODELS to MODELS_OLD and RADIO to RADIO_OLD to ensure that I have a "clean" Zorro - so no models or settings are involved.

I used this firmware:
[zorro-edgetx-v2.7.1.zip](https://github.com/EdgeTX/edgetx/files/9943239/zorro-edgetx-v2.7.1.zip)

After using the trim buttons on ETX 2.7.1, the TX crashes, reboots and then shows that sign:

[image]

msX99, 2022-11-05:
I tried again with zorro-a55aff0.bin from edgetx-firmware-v2.7.1.zip and now it works. I don't know what exactly happend, but probably I used a wrong firmware.

closed 2022-11-05 by msX99

pfeerick, 2022-11-05:
The firmware you uploaded before was for the zorro also, but there were a couple of bytes different to the zorro-a55aff0.bin, so maybe a corrupted download?

msX99, 2022-11-10:
I have created a new feature request (#2705) to avoid corrupted firmware flashing.