EdgeTX/edgetx #2674: EL18 Production switch default positions need adjustment.

richardclli, 2022-11-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EL18 production will have modified switches arrangement.  Need to adjust code for default settings.

Current default settings:
SWC is toggle
SWE is toggle

### Expected Behavior

Expected EL18 production default settings:

SWC changed to 3 pos
SWE changed to 2pos

### Steps To Reproduce

1. Use empty settings SD card
2. Switch on radio

### Version

Nightly (Please give date/commit below)

### Transmitter

Flysky EL18

richardclli, 2022-11-04:
How to work this out is a problem.  Current main do not have EL18 build options, so make the change only in 2.8 branch?

gagarinlg, 2022-11-06:
The software already knows if it is a NV14 or a EL18. When @raphaelcoeffic I/O changes are added, we should be able to find a way to automate this.

raphaelcoeffic, 2022-11-06:
I was thinking the other day, it would be nice to have some "factory settings" stored on some hidden storage (obviously not the SD card...). They could be made to match the version shipped (like internal module, switch types configuration, etc).

gagarinlg, 2022-11-06:
The SPI flash? Some eeprom?

pfeerick, 2022-11-07:
This is a similar idea to that that one of the SBC manufacturers had, since it was not possible to just stick any Linux image on a SD card at boot - need a board-specific boot image. So they started talking about using SPI Flash to store the uboot bootloader which would then boot the SD card... and thus get around this. Seems to have completely flopped, though, as the other manufacturers aren't doing this. 

If it is put into SPI flash or EEPROM, I think that is adding an extra step for manufacturers - so the question then needs to be "will they actually do that or are we wasting our time on that feature?". Another option could be to use the SD card, but add another file (perhaps in RADIO, since that folder isn't doing much else? :grin:) and have all the factory settings there. Thus it would be loaded as part of the manufacturer provisioning of the card. And can be retrieved from the manufacturer's website if needed.

raphaelcoeffic, 2022-11-08:
Right, maybe it could on SD card. I think we discussed having some sort of factory reset earlier. Newbies tend to configure all sorts of things and then be confused how to get back.

pfeerick, 2022-11-08:
Plus, there's always the "oh crumbs, I stuffed it up big time... how do I reset it" moments ;) :grin: Don't have to be a newbie for that to happen ;)

PR #2675 (2022-11-05): chore(el18): Change default switch configuration to match production

Fixes #2674

Summary of changes:
closed 2022-11-15 by pfeerick