EdgeTX/edgetx #2679: Horus X12S 6 position switch icon

Kevltan, 2022-11-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Screen refresh fail for the 6 position switch icon  on screen X12S?

### Expected Behavior

Switch moved to positions 1,2,3,4,5,6 but screen does not respond until refreshed by touching screen. The 6 position switch responds in channel monitor instantly when selected.
Using firmware 2.8 self build but is also reproduced in 2.8 RC3.

### Steps To Reproduce

Rotate 6 position knob to any position no response on screen of movement until screen touched/refreshed.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

eshifri, 2022-11-05:
The same in TX16S simulator

eshifri, 2022-11-05:
If I read the code correctly the problem is here:
https://github.com/EdgeTX/edgetx/blob/579758a22148738982505d11d252b4da72cdff12/radio/src/gui/colorlcd/layouts/sliders.h#L50
value == newValue always.
If I create a separate checkEvents() for MainView6POS and invlidate() it always the screen is redrawn.
I was about to submit a PR, but went to the calibration screen to see what is going on there. ^POS calibration does not work at all now (at least for simulator).
I remember there was plan to remove it from the calibrated analogs, but it is still present on the calibration screen.
After unsuccessful calibration it becomes unusable (at least in the simulator).

Luznatural, 2022-11-05:
I have noticed that the way you do the calibration has an influence in the behaviour of the 6 position switch in Frsky X12S, using both ETX 2.7.1 and ETX 2.8.

Let me elaborate a bit:

When you perform a calibration, if you start moving the rotating knob in a counter rotating movement (turning leftwise) from the position number 3 to the 1 and then to the 6, most of the time the calibration only detects 4 positions instead of 6. If you do not perform another calibration, the knob will remain forever with only 4 positions.

But if you start the calibration in a clockwise form (turning rightwise) from position 3 to 6 and then 1, most of the time the calibration sucessfully detects all 6 positions.

It,s strange, isn't it?

eshifri, 2022-11-05:
Jus tried it again in the TX16S simulator - no effect. In both direction it is stuck at one number (3 or 4).
And additional calibrations do not help in 2.8; in 2.7 they did.

pfeerick, 2022-11-05:
Is the 6pos veining centered (or as near to centre as possible)? I'd have
look at the code to be sure, but if it's the same as the pots that would
explain the lopsided behaviour.

Kevltan, 2022-11-05:
I think its a screen refresh problem. Once calibrated "read tedious" and in
normal screen the numbers sometimes don't respond at all or very delayed on
the screen, yet if you touch the screen anywhere they change instantly to
correct value. The same problem exists in calibration but this time you are
unable to touch the screen or else you move into to calibration complete
screen. I think if the screen is refreshed every time the 6 position switch
is indexed the problem will be resolved.

eshifri, 2022-11-05:
@Kevltan I think you are right: the position of the 6POS is not displayed on calibration screen as well now.
@pfeerick I'm starting from position 3.

Kevltan, 2022-11-05:
Yes it is very hit and miss sometimes it moves sometimes it won't move at
all and then you think it's solved and then it reverts. I think it's down
to what's going on in the background programming at the time you enter
either calibration or the main screen. Screen refresh calls maybe the
answer.

eshifri, 2022-11-05:
I have a work-around for screen refresh and it looks like it calibrates for 5 positions. This is the reason the condition here is alwais broken
[edgetx/radio/src/gui/colorlcd/layouts/sliders.h](https://github.com/EdgeTX/edgetx/blob/579758a22148738982505d11d252b4da72cdff12/radio/src/gui/colorlcd/layouts/sliders.h#L50)

Line 50 in [579758a](https://github.com/EdgeTX/edgetx/commit/579758a22148738982505d11d252b4da72cdff12)

eshifri, 2022-11-05:
Actually, for MainView6POS value==newValue==0 (where int16_t newValue = calibratedAnalogs[idx];) always.
This explains problems with screen refresh.
Still it is not clear what is wrong with calibration.

Luznatural, 2022-11-06:
I do not have a touch screen in my frsky X12S, so I cannot test the refreshment as Kevitan does.
I bought the items to make the mod, but they have not jet arrived.
Sorry

eshifri, 2022-11-06:
You do not need a touch screen for this - just press "Enter". The menu will open on the screen, you can press "RTN", the menu will close and the screen will be refreshed.

closed 2022-11-06 by pfeerick

Kevltan, 2022-11-06:
Thank you! That works great have tested and solves the problem in
calibration and main screen 🙂👍👍

Luznatural, 2022-11-06:
Thanks to all. That was quick!!!

Kevltan, 2022-11-06:
Hi, I had problems when I initially did the mod with the touch being
inverted but there solutions elswhere using reset_gt911 to reset to zero
using a CLI editor in Windows. The other problem I had was cable routing
for the SDL/CLA bus line wires. I had L1 and LS sliders jittering and slow
to respond on the screen. I re-routed the wires away from the back L1, L2
sliders and the problem went away. Think these wires may have been acting
as antennas and picking up stray RF from the board (the RF module is not
screened like in the TX16S). I have since added a 4mm diameter screened 2
core cable with outer screen wire (heatsrunk) soldered to right hand lower
pad on the IMU which is at 0v potential. No problems since.

Luznatural, 2022-11-07:
Hi Kevltan,
Thanks for your hints, that I will take seriously in account.
Would it be posible that you send me some pictures of the way you routed the cables and the shielded cables themselves?
I do not understand the part.where you use reset_gt911 to reset to Zero. Would you be so Kind to elaborate a little bit?
The original mod is described in German. Unfortunly, I do not speak it, and Google translator is not very good for.the details. So, I have to take over difficulties.
Thank you very much in advance.
Very Respectfully

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

Kevltan, 2022-11-07:
Wish I had taken some pictures now! Will try to get some done. My android
phone automatically converts the German to English so unless you speak
another language it should do this automatically for you. I think there is
a new part added now to German right up that tells you how to reset your
screen if you need to - you may not need to. Do not use HIS firmware
version of 2.8 use one off Github something like 2.8 RC3 this has the
correct touch screen orientation for you touch panel if you HIS there is
good chance it will invert the touch panel - this is OK if you use his
firmware but not if you use the latest firmware off Github.

Kevltan, 2022-11-21:
See discussion ticket #2771

PR #2684 (2022-11-06): fix(color): 6POS display on the main and calib. screens (#2679)

Fixes #2679 

Summary of changes: add checkEvents() to the MainView6POS.

This fixies display issue, but it looks like 6POS calibration is broken.