EdgeTX/edgetx #2686: "Copy trims to subtrim is confusing"

gjhaynes, 2022-11-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On the Output screen the button "Add all Trims to Subtrims" works in a completely logical manner.
However, the context-sensitive menu for a specific channel departs from logic (IMO). "Copy trims to subtrim" behaves additively and does not reset the trim lever to centre. This can be confusing since every press of this menu option repeats the addition of the current trim lever setting. 

### Expected Behavior

The "Copy trims to subtrim" menu item for each individual channel should behave in the same way as the "Add all Trims to Subtrims" button, but only for the selected channel. The trim lever should be reset to centre and the menu text should also say "Add trim to subtrim"

### Steps To Reproduce

1. View the Output screen
2. Centre all trims and reset all subtrims to zero
3. Use a trim lever to apply +10% trim to channel 1, then press "Add all Trims to Subtrims". 
4. The Subtrim value is set to +10% and the trim lever position is reset to centre. All good.
5. Add another +5% trim to channel 1. The trim setting displays 15% (correct).
6. Being happy with the trim condition for channel 1, tap on its channel line and press "Copy trims to subtrim".
7. The trim setting is now set to +20%, but we wanted to save the 15% value. Not good!

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Firmware is 3.0 RC4

pfeerick, 2022-11-07:
We've had discussions before about the phasing of the two - the global "Add all Trims to Subtrims" which does a copy **_and_** reset, and the individual "Copy trims to subtrim", which only copies that individual channels trim to subtrim, with no satisfactory resolution. 

I think I suggested "Move trims to Subtrims" to try and make it more clear that Copy would only copy, whereas the "Add" would also reset the trims afterwards, but then some other interaction, but then some other aspect came into play. But it was suggested "Add" was more intuitive since the channels keep getting added too, every time you press the button, and move the trims again. 

I think the general consensus in that discussion, and other mentions I have seen, was it was felt the copy menu should also reset (or at least prompt) the trims, but we also didn't fully know *why* it was done that way, so left it alone for now. But this page has been reworked since, so now is the time to make those sorts of changes.

gjhaynes, 2022-11-07:
Thanks Peter. An enhancement that would be really cool is to provide a function that does exactly what the "Add all Trims to Subtrims" button does. Then I could use a momentary switch to perform that magic while in flight. Trying to do it in flight using the Output screen is not really a good idea 😉. I'm aware of the "Inst trim" function, but being able to transfer trim settings to subtrims, rather than stick positions to subtrims, is far more practical when trimming an aircraft in flight. 
Cheers,
Geoff

raphaelcoeffic, 2022-11-07:
@gjhaynes just so you know, we have to stop the mixer while performing this magic: you might not want to do this while flying ;-)

gjhaynes, 2022-11-07:
@raphaelcoeffic Aahh. That is an important piece of information! Do you have a rough estimate of how long the mixer is stopped while performing the trim transfer? Does this risk apply to the "Inst. trim" function? There's nothing stopping anyone using this in flight via a special function.
With that caveat, I still think the suggested function has merit as a shortcut way to invoke "Add all Trims to Subtrims" after trimming and landing the aircraft.

pfeerick, 2022-11-07:
I don't think there is any issue with instant trim - it really just alters the trims, rather than altering the output mixing for multiple channels. After all, instant trim is *meant* to be used in flight ;) 

To do what you want, IMO you should be using the two steps of instant trim for in-flight adjustment, and then moving the trims to subtrim (if you refuse/are unable to correct it mechanically) once on the ground. Given it's only a few button presses away (and even at the top of the page now), I don't see the value in making the "Add all Trims to Subtrims" any more accessible than it is (ignoring for a moment the inherent danger) - how often is this going to be used by a majority of users? But that's just me.

gjhaynes, 2022-11-08:
Maybe it's just me Peter, but I find it quite a challenge trying to adjust the aircraft trim in flight by holding the stick perfectly stationary in a given position while simultaneously pressing a button for Instant Trim. If Instant Trim captured the trim lever setting, which has fine granularity and retains its value after each click, rather than capturing the gimbal position with relatively coarse granularity and reliant on a very steady hand, I would consider it useful. As it is, it offers no advantage (IMO) over trimming in flight with the trim levers, then making the appropriate subtrim adjustments on the ground with "Add all Trims to Subtrims" (or mechanical adjustment as required). Am I missing some advantage of Instant Trim over the use of the trim levers? It wouldn't be the first time :)
At least I now understand the underlying technical reason for not wanting to mess with output mixing in flight. Thanks.

gjhaynes, 2022-11-10:
I'm having trouble understanding how holding the sticks very steady while triggering a function is better/easier than simply using the trim levers to trim out an aircraft in flight. It's a trivial task to move the trims to subtrims after the flight with your cool "Add all Trims to Subtrims" button.

MervDale, 2022-11-10:
> I'm having trouble understanding how holding the sticks very steady while triggering a function is better/easier than simply using the trim levers to trim out an aircraft in flight. It's a trivial task to move the trims to subtrims after the flight with your cool "Add all Trims to Subtrims" button.

That is because instant trim once activated grabs all your stick positions  "when you are in steady flight" ( rudder, elevator, aileron) and instantly locks them to your trims instead of you adjusting each trim individually. I have it setup on my momentary switch along with timer reset using logical switches and special functions  ;-Q