EdgeTX/edgetx #2691: Expand the functionality of Trainer mode

gjhaynes, 2022-11-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Allow specific channels to be under student control in Trainer mode

### Describe the solution you'd like

I'm sure I am not the only instructor who finds it very useful to restrict student control to a subset of available channels in their early stage of learning. This option has been a standard feature of Spe...um and Fu...ba radios for many years. The ability to introduce the student to the effects of just one or two control surfaces during a lesson can speed up the learning process. As they become more competent, more channels can be assigned to student control.
This functionality would require an expansion of the trainer options in EdgeTX, using checkboxes alongside each channel or stick. Only the checked items would be under slave radio control when the instructor hands control to the student. The remainder are controlled by the instructor.
Giving EdgeTX the same trainer flexibility as competitive products will increase the adoption of EdgeTX-equipped transmitters by clubs who play an important role in attracting new members to the RC hobby and teaching them to fly.

gjhaynes, 2022-11-08:
Trainer setup screenshot from a popular brand of Tx. Each channel can be enabled/disabled for student control:

[image]

pfeerick, 2022-11-08:
While I won't by any means say OTX/ETX has as simple an interface 🤭, I think you'll find it's possible to do this, and much more... 

i.e. I think you should be able to do this at a system/global level, by changing any of the four sticks from `:=` (replace) mix to `OFF` in the Radio Settings -> Trainer. 

And at a model level, you should be able to do this when configuring the trainer SF  - there are options to either pass over the 4 sticks, individual sticks, or channels (which is basically custom - _you_ choose which TR channels are assigned to what mixer channels). You can also use the trainer SF multiple times, so you could, for example, say you just want the Ail and the Elev sticks with two separate lines.

dlktdr2, 2022-11-08:
Although it's a bit more work than checking a box you can already do this in EdgeTx. Say you want to only allow the student to control the Throttle (Ch1)

- Under your mixer page, add a new line on Ch1, 
- Switch to your trainer switch.
- Multiplex = Replace
- Source = TR1 (Trainer input 1)

You can rinse and repeat for all the channels you want the student to be able to control. If you do this way, disable the Special Function Trainer. 

But under the special function Trainer option, would be a nice GUI addition for simple checkbox selector.

Eldenroot, 2022-11-08:
I recognize this screen from Spektrum :) yeah, would be nice

gjhaynes, 2022-11-08:
If you compare the suggested options in EdgeTX with the Spektrum screen, can you imagine how much simpler the latter is to set up any combination of student-controlled channels during a training session? I have both a TX16S Mk2 and a Spektrum DX9 and I can tell you which one is preferred by the average instructor. Expanding the trainer function to use checkboxes for channels would most likely reverse that preference.
I'm not expecting any priority for this enhancement, but it would be nice to see it flagged for implementation in a future version.  Pete, maybe you could put it into one of your nighties 🤣 🤣 
BTW, the TX16S is my favourite Tx hands down! But I'm a techie and I love the power of ETX.

sande005, 2022-11-09:
How is this wish different in practice than either:
1. The specific Special Functions that allow each of the 4 primary trainer controls to be turned on/off with a switch?  I currently have a switch that will turn on/of from the Master the Student throttle and/or rudder.  When off, Master has control all the time.  With logical switches or mixing, pretty complex fine tuned controls can be accomplished.  Mine can give control, or not, even in flight.  And force the student to always be in low rates, but when I get control rates are wherever I wan them to be based on my switches.
2. Ditto when using trainer channels 1-8 as Sources

gjhaynes, 2022-11-09:
@sande005 Maybe I should have titled the request "Simplify the Setup of Trainer Mode".

pfeerick, 2022-11-09:
It's not so much the title, but knowing from what level of knowledge of a
system the suggestion is made. I. E. If you are fully aware of these
options, and make the suggestion, that has more meaning than done one made
by someone who doesn't (since they didn't even know half the options
existed/don't understand how the system works).

This is the sort of QoL improvement we will need to consider in the future,
as if you stop and look for a moment, special functions are basically all
the same - what function, what trigger, one or maybe two parameters, and
sometimes and enable checkbox. So to have more options, some major changes
may be needed behind the scenes... Including variable number of parameters.
So at some point GF/SF is going to need an overhaul so it can be more
flexible.

gjhaynes, 2022-11-10:
In summary, I'm suggesting that the Trainer setup could be similar to other basic features like Timers.
[image]

sande005, 2022-11-10:
Clear that you would like a simpler interface for setting it up, which is appealing.  But the flexibility can get lost.  For example, I have two switches on the master as "sub functions" when the trainer switch is held - one turns on/off throttle for the student, one does same for rudder, while in the air.  Also, the scaling and centering is lost - ie Futaba has very different throw amounts from Spectrum, which both are different than from OpenTX/Edge.  So tuning of the buddy box input would be lost or at least restricted to one brand of radio.  Only use a wired connection so can't speak to use of a MPM and what impacts the suggestion may or may not have.

pfeerick, 2022-11-11:
I would be more inclined to have an intermediate screen/option ... almost like a "basic" / "advanced", or "simple" / "classic" ... with this being the "simple" and then when you press classic/advanced/whatever it just behaves like it always has. Or maybe the order of the options is wrong - and the switch should be first...  i.e. configuring that would then show the other elements, also putting you into that mode.  And this is the right place to put this IMO, as the `master/jack` option makes no further use of the screen space, and really should give you some trainer options. I think this is also where what was brought up here #2704 is also relevant, as this would be the right place to allow for the selection of model-specific profiles...

ChrisOhara57, 2023-01-04:
Fyi I just added a information in #2704 how many of the request can be fulfilled at the moment with OTX and edgeTx as it is by using **input line** for mixing. Which seems to be rather unknown to community.

tippfehlr, 2023-07-22:
I also want to give the slave command over channels beyond the sticks. I have no problem configuring them in model settings, but I didn't find it at first. I assumed that the trainer page is the only way and it was limited to 4 channels.

TLDR; I think it should be mentioned (at least in the wiki) that there are two ways to configure the trainer

pfeerick, 2023-07-23:
> TLDR; I think it should be mentioned (at least in the wiki) that there are two ways to configure the trainer

@phileville Something to mention in the manual? I know it is briefly mentioned (e.g. https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/model-settings/special-functions) but perhaps more detail is needed. I think you may have mentioned this already, but probably need to give some thought as to how to have another general category for topics/themes that bridge across multiple parts of the radio... i.e. Trainer can involve radio setup, model setup, special function, mixes ... so can be a bit messy trying to bring it all together and explain all the possibilities.

phileville, 2023-07-24:
@pfeerick - I agree that this should be covered in-depth with a how-to article.. It is already on my to-do liet, but I will move it a bit further to the top.  In addition, I think that this area could really benefit from a re-look from fresh eyes at a better way to configure trainer options... also keeping in mind those that use it as more of a PPM passthrough mode for using two controllers to control one complex model.