EdgeTX/edgetx #2694: XLite Pro Gyro support

Gerd-St, 2022-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I updated my X-Lite Pro from OpenTX to EdgeTX 2.8 RC4 and converted my models using EdgeTX Companion. All went fine except for one simple car model where I am using GyrY for steering.
Within companion all looked ok, but after downloading to the radio I got an empty model for this car there, without mixes and so without any channel output.
So I had to add the mixes directly at the radio to get the model working.
So there seems to be a bug in Companion preventing the model from being downloaded to the radio correctly.

[image]
[X-Lite Pro 2022-11-07.zip](https://github.com/EdgeTX/edgetx/files/9959868/X-Lite.Pro.2022-11-07.zip)

### Expected Behavior

Identical model settings between Companion and radio.

### Steps To Reproduce

- Load the model definitions (zip file) into Companion
- Download to the radio
- Check the mixes at the radio if they are there

### Version

Other (Please specify below)

### Transmitter

FrSky X-Lite / S / Pro

elecpower, 2022-11-13:
Confirmed

elecpower, 2022-11-13:
After more digging there are both yaml and ui differences between Companion and radio (using sim).
[image]
[image]

Companion yaml mapping
>  } else if (IS_TARANIS_XLITE(board)) {
    tbl.insert(tbl.end(), {
                              {"S1", "POT1"},
                              {"S2", "POT2"},
                              {"GyrX", "GYRO1"},
                              {"GyrY", "GYRO2"},
                          });

Radio yaml mapping:
> const struct YamlIdStr enum_MixSources[] = {
  {  MIXSRC_NONE, "NONE"  },
  {  MIXSRC_Rud, "Rud"  },
  {  MIXSRC_Ele, "Ele"  },
  {  MIXSRC_Thr, "Thr"  },
  {  MIXSRC_Ail, "Ail"  },
  {  MIXSRC_POT1, "POT1"  },
  {  MIXSRC_POT2, "POT2"  },
  {  MIXSRC_TILT_X, "TILT_X"  },
  {  MIXSRC_TILT_Y, "TILT_Y"  },

The FrSky manual https://www.frsky-rc.com/wp-content/uploads/Downloads/Manual/X-Lite%20Pro%20S/Taranis%20X-Lite%20S%20Pro-%20Manual.pdf has a radio screen shot referring to GyrX.

I also note the manual has a screen shot with Gyro setup parameters which EdgeTX radio and Companion do not have as configurable.

So which side needs changing? I'm leaning towards the radio for mix and both if setup needs adding.

Gerd-St, 2022-11-13:
There is a further problem concerning the gyro, see also my bug report #2695  which is actually still open.

pfeerick, 2022-11-13:
Radio is correct, it was changed midway through 2.8 IIRC, but obviously not
done  in companion.

pfeerick, 2022-11-15:
@Gerd-St would you be able to try the updated Companion build out @ https://github.com/EdgeTX/edgetx/pull/2740/checks (pick the version of companion you need on the left side, and then when you scroll to the bottom there will be an "Artifacts" heading, and under that is the link for the zip file with the installer/package.

Gerd-St, 2022-11-15:
I just tested with the Ubuntu version and it works as should now. I opened my last backup taken by OpenTX Companion and imported it frshly into the "Artifacts" build, then downloaded to the X-Lite Pro and checked the "Buggy" model settings for correctness, including a test drive with the model.

I am wondering about my second bug report #174 concerning the X-Lite Pro internal gyro dampening which is still on "triage". Would be great if this could be resolved before finally releasing version 2.8, too.

pfeerick, 2022-11-15:
Triage means not responded to or not verified yet. I believe there was some tuning done in an earlier PR, but I don't have this TX or if it was over-dampened. @raphaelcoeffic Can you comment on this over in #2695?

PR #2740 (2022-11-15): cpn(fix): gyro and joystick
Fixes #2694

Summary of changes:
- align gyro and joysticks for X10, X12S and X Lite to radios
- add yaml conversion for GYRO1 and GYRO2 to TILT-X and TILT-Y
- add bin conversion
closed 2022-11-15 by pfeerick