EdgeTX/edgetx #2695: Heavily dampend gyro sensor on FrSky X-Lite Pro

Gerd-St, 2022-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am using the GyrY sensor for steering a car in OpenTX. After converting to EdgeTX 2.8 RC4 the sensor still works, but reaction is extremely slow and no longer usable for steering.
[image]

### Expected Behavior

Quick sensor response as with e.g potentiometers or sticks.

### Steps To Reproduce

- Make a simple model with just mixes as shown in the screenshot
- Test the gyro function using the channel monitor and check for responsitivity.

### Version

Other (Please specify below)

### Transmitter

FrSky X-Lite / S / Pro

raphaelcoeffic, 2023-06-12:
The algo has changed to adapt to multiple other sensors. The issue with the X-lite Pro is that none of the devs has this radio and thus cannot really test anything. If you want a higher sensitivity, you can still use the output with a factor of over 100% (ex: 200%). It is sometimes hard to find a proper balance between stability (filtering jitter out) and reactivity. We might have to try it ourselves and find a way to make it more adaptable / tunable.

Gerd-St, 2023-06-12:
The gyro worked as should while I was with OpenTX 2.3.x. So you can still use the dampening method from there if the new on in EdgeTX is not working reliably. And please also consider my comment on issue #3497 from Apr 20 where the gyro does no longer work at all after adding a further mixer line, so issue 3497 is no longer just a duplicate of this one. Sensity is not the problem by the way, it is the delay.