EdgeTX/edgetx #2696: Duplicate Models are back

planemann2000, 2022-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.8.0 RC4 flashed; now getting duplicate models on powering up radio then deleting duplicate models in Model Select page, then open up companion, now other duplicate models will be in companion, and will migrate to back to the transmitter after companion is closed.  

### Expected Behavior

No duplicate models

### Steps To Reproduce

1 Start up transmitter
2 Go to Model Select page Delete duplicate models
3 Start Companion, more duplicate models,  2.8.0 version of companion
4 Close companion, go back to Model Select page the duplicate models will be there from compainion

Can duplicate bug with both my transmitters.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

dlktdr, 2022-11-08:
Can you send your model and radio folders again.

Is it possible to upload a video of how you are getting this to happen?

planemann2000, 2022-11-08:
[radio and model files.zip](https://github.com/EdgeTX/edgetx/files/9964402/radio.and.model.files.zip)

planemann2000, 2022-11-08:
I'm not 100% certain but, I may have found the problem. I went back to RC3 and still had dup. on companion that transferred to the radio. I changed out USB cable and haven't been able to duplicate the problem since. I will keep watch for now

planemann2000, 2022-11-08:
Do you want me to close this issue out for now?

dlktdr, 2022-11-08:
No you can leave it open, a faulty USB cable still shouldn't do that. I would still be interested to see in a video what is happening with the bad cable. It shouldn't ever make a duplicate, without you telling it to. Then I can try to repeat it and find the root cause.

Should have some time to do testing tonight.

planemann2000, 2022-11-08:
You are right; reflashed RC4 and checked for dup. on radio (none), plugged in with old cable opened companion and had a dup. closed com. disconnected from computer and the dup. was in Model select.

planemann2000, 2022-11-09:
I am thinking the issue is with 2.8.0 rc1 companion. The dup. is created after I make a change to a model in companion, write the change to the radio, back out of companion (doesn't matter whether I save or discard the change) the resart companion. I reverted back to 2.7.1 on my main tx16s tried to duplicate the steps in 2.7.1 companion as with 2.8.0 com. with no issues as of now. I checked for an updated (rc3) companion and have uninstalled and reinstalled the 2.8.0 app with the duplicates reappearing after newly installing.    Maybe that will help.  No video Sorry

planemann2000, 2022-11-09:
I did find the 2.8.0.rc4 companion download after a lot of searching, github is not the easiest web site to navigate in my opinion. So far all is good. Will do more testing. Thanks

planemann2000, 2022-11-10:
I give up, getting duplicate models after a change in companion, after shutting companion down, disconnecting from the computer (windows 11) restarting companion and usually 1 to 2 duplicate models in companion which transfers to the radio model select page. And it isn't 100% repeatable.

dlktdr, 2022-11-15:
Sorry for delay.

I did found an issue with duplicates showing up in companion. If you try to drag a model in the current editor it duplicates it. It should really disable dropping from itself. Should only allow another window.

Could it be this?

closed 2022-11-22 by planemann2000