EdgeTX/edgetx #2697: "Show Elapsed" timer option does not work on RM TX16s

phileville, 2022-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When selecting the "Show Elapsed" option in Timer1 on TX16s, it shows remaining time, like "Show Remaining" is selected.  On the Companion it works like it should.  On the RM TX16s it does nothing.

### Expected Behavior

I expect that when the "Show Elapsed" option is set on the timer that the timer starts at 0 and counts up until the designated value and beeps and turns red once value is exceeded.

### Steps To Reproduce

1. Configure Timer1  with a start time of :30
2. for the direct option, select "show remaining"
3. extit out of model setting and make time start
4. Notice that it starts at 30 and counts down.  It should start at 0 and count up.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Version 2.8.0 RC4

eshifri, 2022-11-09:
Your description (steps to reproduce) is confusing: if you select "Show remaining" the timer starts from 0:30 and counts to 0. This is correct behavior.
If you select "Show Elapsed" it will switch from 0:30 to 0 when countdown starts and will end at 0:30.
Audio announcements will follow the selection if you use "announce minutes". Seconds are always announced for "remaining" to avoid confusion: the problem is if you chose 10s it will announce every second for first 10 seconds and than switch to every 10s announcement, which is, probably, not you want.

phileville, 2022-11-09:
Sorry, the recreate steps were backward.  Set the timer to show elapsed and it still will count down from 30.  It is important that you test it in the radio, as the simulatir works correctly.

eshifri, 2022-11-09:
I cannot reproduce it: it works correctly for me on TX and in the simulator.
Announced time is "remaining" for seconds, but the display is correct.
Can you please check again?

Richrude19, 2022-11-09:
on the model I looked at I have timer widget in the top banner and in the main body of one page if I change it to elapsed the main body resets to 5:00 minutes changes to zero when it starts to time. the banner starts at 5:00 and counts down. Does it on both of my radios,

pfeerick, 2022-11-09:
At first I thought I could reproduce this, but it also works for me. The only 'niggle' I have with the built-in "Timer" widget is when set to "Show Elapsed", it will initially start at "00:30" (with a 30-second timer), and then when activated will start counting correctly. I don't know whether this is good or bad - I guess you know how long the timer is set for, but just seems a bit odd that it's not 00:00 if it's counting elapsed and hasn't started counting yet. 

Can I get a confirmation from both @phileville and @Richrude19 that you are using the "Timer" widget when this misbehaves for you, and not "Timer2" (which is a Lua widget, and hasn't been updated yet to support the "show elapsed" option). As I can't repoduce this with the built-in Timer widget on my TX16S, but Timer2 will always countdown (show remaining).

Richrude19, 2022-11-09:
Using timer 1 both  displays from TELE key setup

phileville, 2022-11-09:
I will double check this evening when I am back home and let you know, might be the widget issue (Timer2) that Peter pointed out. I don't know for sure which timer I used...

Richrude19, 2022-11-09:
https://user-images.githubusercontent.com/111724494/200808928-38e51b84-3f81-4e69-80a9-414c78b601cd.mov

https://user-images.githubusercontent.com/111724494/200808986-168c2774-6c53-4554-a679-94ddc86880eb.mov

hope one works

eshifri, 2022-11-09:
@pfeerick it starts from 0:30 on purpose: I think one wants to know the timer setting before the start.
@Richrude19 I think the top bar timer is not affected by this setting. At least I have never tried. :-)

phileville, 2022-11-09:
OK Guys, I figured out what the issue is, it was how I had it configured.  "Timer" widget does work like it is supposed to - both in the main screen or topbar.  "Timer2" does not.  However, my issue was that I set a value widget with the value being "timer1"... Of course in this case, the value widget does not take into consideration the whole show elapsed vs show remaining.  Sorry to waste everyone's time with this...but it is good to know how it works.

closed 2022-11-09 by phileville