EdgeTX/edgetx #2699: System (TX Control) Buttons not working/correctly

oldnansome, 2022-11-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

New Radiomaster16 TXs up graded to Edgetx.
No touch screen and the buttons are out of sequence or not working.
The sys button opens radio settings.
The Return button does nothing
The Model  button acts as the Return button
Page Up moves long the toolbar folders
Page Down does nothing.

### Expected Behavior

Touch screen to work and buttons to work as designated

### Steps To Reproduce

From turning on

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

gagarinlg, 2022-11-08:
Please open your radio and check all connectors. This is a hardware issue

oldnansome, 2022-11-09:
I have gone back to the original SD card (Opentx) that came with the radio. Loaded that bootloader, bearing in mind it is the same as the one i used to flash EDGETX onto the transmitter. Turned it off then turned back on. The transmitter still has edge tx and now all the buttons work!
Put the flashed sd card (edgetx 2.7) back in and this works, as well as giving me the protocols missing from the original sd card. I am wondering whether it was a corrupted Bootloader or whether the sd card at 16G was too big as the original card is only 256Mb.

closed 2022-11-09 by oldnansome

pfeerick, 2022-11-09:
16GB and larger SD cards are fine, however, you may have issues with some larger cards which typically show up as intermittent very long power-on times (as in 30-40 seconds or more). There was a patch to help prevent this, however, you will need 2.8.0 to benefit from that. 

Anyway, it sounds like something went wrong the first time you flashed the firmware, and when you re-did the firmware flash, the issue was resolved?