EdgeTX/edgetx #2706: Servos Not working

oldnansome, 2022-11-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Have now set up Edge Tx on RMtx16s. (Hopefully)
Using DSM protocol and have bound spektrum receiver AR636sx. TX set for TAER. 
Throttle servo works but the others do not. TX monitor shows stick movement. The servos all work when tested.

### Expected Behavior

Expect servos to work with correct stick input

### Steps To Reproduce

If I knew I wouldnt be here

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

ParkerEde, 2022-11-10:
in my experience, it is always due to the "disable channel mapping" hook, which is not set by default. But if you have created the outputs and the mixers all according to the Spektrum mapping, the check mark should be set as well.
I fly e.g. the Blade 130S with an ext. MPM and everything works as it should. Therefore, I can not really imagine a problem with EdgeTX itself.

oldnansome, 2022-11-12:
Have sorted this by reboot and channel mapping

closed 2022-11-12 by oldnansome