EdgeTX/edgetx #2713: MPM protocol options missing in various protocol

richardclli, 2022-11-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

e.g OMP protocol uses CC2500 but do not have freq. tuning options.

Accordingly to Pascallanger's description:

EdgeTX is pulling at startup all the protocols and their features. But at that time and until the protocol is actually launched there are some unknowns.
Once the protocol is launched, multi sends in each status frame to the radio what exactly should be displayed. -> This is where the issue is. EdgeTX seems to ignore the changes being sent.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/747

### Expected Behavior

e.g. OMP protocol should shown freq. tuning options.

### Steps To Reproduce

Internal Module -> OMP protocol 
No freq. tuning options

### Version

Nightly (Please give date/commit below)

### Transmitter

BetaFPV LiteRadio3 Pro, iFlight Commando8, Flysky EL18, Flysky NV14, FrSky Q X7 / Q X7S (ACCST), FrSky Q X7 / Q X7S (ACCESS), FrSky X9D, FrSky X9D+, FrSky X9D+2019, FrSky X9E / X9E Hall, FrSky X9 Lite / Lite S, FrSky X10 / X10S (ACCST), FrSky X10 Express / X10S Express (ACCESS), FrSky X12, FrSky X-Lite / S / Pro, Jumper T12, Jumper T16, Jumper T18, Jumper T-Lite, Jumper T-Pro, Radiomaster T8, RadioMaster TX12 / TX12MK2, Radiomaster TX16S / TX16SMK2, Radiomaster Zorro

richardclli, 2023-01-03:
@raphaelcoeffic I think you may need to follow this up.  For testing, all NRF24L01 and CC2500 protocols should be able to fine tune frequency.  You can use OMP protocol to test.

richardclli, 2023-04-18:
This is similar to the problem in Flysky options.  Information given by MPM need to be refreshed in the UI for setting up the options.  @philmoz Can you look into this as well?

richardclli, 2023-04-18:
Issue #3046 is related to this as well.

mha1, 2023-04-20:
@richardclli @pfeerick I was browsing through issues tagged 2.9 and found this one. It originally talked about OMP missing frequency tuning and that in general all NRF24L01 and CC2500 protocols should have the frequency tuning option.

The bible here is MPM's [Multi_Protos.ino](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multi_Protos.ino) which defines all protocols properties. All CC2500 protocols are defined with RFTUNE, but only a few NRF24L01 protocols are tagged with RFTUNE (GD00X, KF606, Q90C, SLT, V911s, XK). The NRF24L01 protocol OMP is **not** tagged with RFTUNE. Hence EdgeTX is correct in not displaying the RF tuning option.

Of course there is a mismatch between Multi_Protocol.ino's OMP definition and MPM's documentation which lists OMP as tunable protocol. This needs to be addressed with MPM. If MPM is updated to allow RF tuning EdgeTX will show the option. I have issued MPM https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/848 for clarification.

If #3046 is still valid maybe @philmoz can do his magic.

As #3369 fixed the problem with protocol XK I think this issue should be closed.

richardclli, 2023-04-20:
Please refer to MPM#747 Issue that I included in the description.  You missed the important discussion in that issue.  The problem is not in the definition.  Pascal changed it intentionally because the emulation layer can choose which chip to use and depends of the chip the set of options will be different.  The real problem is the UI is not refreshed according to the return packet given by the MPM, which is the problem of EdgeTX.  You are repeating what I questioned about a few months ago.  ^.^

mha1, 2023-04-20:
ok, sorry, I really didn't know about this discussion. You'd have to update the scanned protocols after selection.

Pascal didn't answer your last question, did he?

richardclli, 2023-04-20:
> ok, sorry, I really didn't know about this discussion. You'd have to update the scanned protocols after selection.
> 
> Pascal didn't answer your last question, did he?

@raphaelcoeffic said he will look into this, but well he is just too busy, and we can help.  I think I will follow this with the help from @philmoz, seems he is quite familiar with the UI.

pfeerick, 2023-04-21:
The UI side is easy - what is needed is the mechanism(s) to track and update the data structures on the radio side when the MPM/AFHDS3 modules change things on their side. Then UI can refresh that once the "update me" flag is raised.

philmoz, 2023-04-21:
> AFHDS3 already has the code to update module data, just need to notify the refresh on the UI side. And I think MPM should not hard as well, just replace the information loaded during MPM init will get the job done.

Can you tell me which file and function this happens in so I can ty and understand the flow?

pfeerick, 2023-04-21:
> AFHDS3 already has the code to update module data

Point to the code please ;)

richardclli, 2023-04-21:
Here is where to process the MPM status packet, has a timestamp already, not sure if this related to UI refresh.

https://github.com/EdgeTX/edgetx/blob/e1fe7fce892bb9c56134be01beb119dcb3d80d43/radio/src/telemetry/multi.cpp#L205

pfeerick, 2023-04-21:
So it sounds basically like we need a function that keeps an eye on the status packets, and if the options are changed from those of the initial protocol scan, that needs to be updated, and the flag raised so the UI knows to refresh?

richardclli, 2023-04-21:
Need to code something and try it out.

mha1, 2023-04-21:
Please check out PR #3504. Tested ok TX16s ok with MPM/OPM.

The culprit is a minor problem in `radio\src\gui\colorlcd\mpm_settings.cpp`. The routine used to fetch the title string for rf tuning (which is used to decide if the rf tune and rssi line is to be displayed or not) just checks the data structure that is initialized after a scan not taking in account the MPM udated status. b&w radios already used the correct function getMultiOptionTitle() which checks if there is a valid MPM status. If so, the MPM updated data is used to return the string based on updated MPM flags. If not, e.g. because there is no MPM connected to the radio or Simulator is running it returns the string based on pre-defined data.

https://github.com/EdgeTX/edgetx/blob/e1fe7fce892bb9c56134be01beb119dcb3d80d43/radio/src/gui/gui_common.cpp#L1298-L1319

richardclli, 2023-04-24:
Nice work, Michael.

mha1, 2023-04-24:
Thanks Richard. Can you please give it a try against #3046. It might fix it too but I don't have hw to confirm.

richardclli, 2023-04-27:
> Thanks Richard. Can you please give it a try against #3046. It might fix it too but I don't have hw to confirm.

I will test this, for sure.

richardclli, 2023-05-07:
> Thanks Richard. Can you please give it a try against #3046. It might fix it too but I don't have hw to confirm.

I have tested it this weekend, however, sometimes it can update the subprotocol after binding, sometimes it cannot.  Need some more time to check what is the root cause.

PR #3504 (2023-05-23): fix: MPM protocol options missing in various protocols
Fixes #2713, fixes #3046 (partly #3500), and fixes #1470

Summary of changes:
- use getMultiOptionTitle() instead of rfproto->getOptionStr() as getMultiOptionTitle() takes updated MPM status in account
- changes to mpm_settings.cpp to check for MPM changing subtype for DSM auto
commit 8893db27 (2023-05-23): fix: MPM protocol options missing for some protocols (#3504)

* fixes #2713

changes:
- use getMultiOptionTitle() instead of rfproto->getOptionStr() as getMultiOptionTitle() takes updated MPM status in account

* code cleanup: removed obsolete functions

* fix #3046

* fixed my complete misunderstanding of how the chosen protocol is updated in Multi DSM auto mode

* removed obsolete check

* update subType list only on rf protocol change

added some comments

* reconsidered implementation utilizing  LV_EVENT_REFRESH to avoid sporadic loss of update

* editorial change

* fix problem where a su [... cut]
closed 2023-05-23 by pfeerick