EdgeTX/edgetx #2714: 2.8 RC4 - LUA widget background() function is not always called when widget is not visible

JimB40, 2022-11-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Widget's backround() funtion is called only when main view is switched to different view.
While any EdgeTX menu or tab is visible widget's background funcion is not called.
So if you for example press MDL to display model configuration neither run() or background() is called

### Expected Behavior

widget's background() function should be called whenever wiget is not visible. so that includes invoking any of EdgeTX tabs. menu etc. that obsucre widget.

### Steps To Reproduce

just run widget on RC4

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2, Other (Please specify below)

JimB40, 2022-11-14:
This will be a bummer I tell you :) 

Did some more test and
- LUA widget's background() IS NOT called when anything from OS is displayed over main view
- LUA widget's background() IS called when main view is different that the one that is used by widget.

So in case two different widget's are using for example crsfTelemetryPop() they still will be racing for data.

Checked Yaapu & iNav Telemetry.

JimB40, 2022-11-14:
Generally LUA background concept is not bad.
Of course any LUA dev can write script that will make problems for OS. But isn't it so for anyt OS?

Stealing buffered data is not to be solved on OS level but two thiings.
1. LUA run in Stand Alone mode shoud have priority - it  should stop all other scripts - as it was designed.
2. LUA  run Widget Mode - should not use any buffered data in background function - just coding practice. (documented)

As for OS  performace
For LUA scripts in Widget Mode there can be certain time-slot in every cycle for execution. If exceeded script is killed or execution is postponed to next cycle.

I know it's not trivial to code (with emphasis on point 3) but IMHO doable.

lshems, 2022-11-14:
Be careful with this one. There was a similar paradigma shift on telemetryscripts in openTX.

Initially, 'background' was NOT called when the 'run' was called and vice versa, as for widgets now. For several (wrong in my opinion) reasons both are now always called. 

So let's keep it simple: when something is shown on the LCD screen, the refresh is called. The background needs to be called explicit from refresh if desired in that case.

Background needs to keep running in all other situations preferably. 

If you want to stop refreshing because some other radio action is done ? Than background should still run.

For me the radio menu pages are allowed to stop background processing as well, but others might object.

One-time script pauses all other lua executions (no calls to run or background, not sure about widget reinits though after stopping?

jfrickmann, 2022-11-14:
Funny, I was troubleshooting a similar problem and was about to file my own issue when I saw that Robert beat me to it.

In my case, the problem is also present in 2.7, so not 2.8 specific. If I add a widget to the top panel, then it should call background when another widget is in fullscreen mode, but it does not do so. For my SoarETX models this means that the battery monitor widget does not run when I fly with score keeping in fullscreen mode - and that can be a serious problem if I drain the flight pack w/o knowing!

To test, load this widget into the top panel. It beeps every 5 sec. in background.
```
-- Widget to test if refresh() or background() are called

local name = "Test"

local options = {
}

local function create(zone, options)
  local widget = {
    zone = zone,
    options = options
  }
  widget.delay = 500
	widget.nxtBeep = 0
  return widget
end

local function update(widget, options)
  widget.options = options
end

local function background(widget)
	local now = getTime()
	if now > widget.nxtBeep then
		widget.nxtBeep = now + widget.delay
		playTone(5000, 100, 0, PLAY_NOW)
	end
end

local function refresh(widget, event, touchState)
	lcd.drawText(widget.zone.w / 2, widget.zone.h / 2, "Test", CENTER + VCENTER + DBLSIZE)
end

return {
  name = name,
  options = options,
  create = create,
  update = update,
  refresh = refresh,
  background = background
}
```

JimB40, 2022-11-15:
So here is like until 2.8

1. One-time scripts - called One-Time scripts or Standad alone 
- init() is called at script start
- run() is called all time.
- works in God mode ie every other script (except Functions scripts (4)  I think) type is halted
- script is stopped and disabled if it misbehaves (e.g. too long runtime, run-time error, or low memory)

2. All-time scripts - called Telemetry scripts for BW, Widget scripts for Color 
- create() is called at widget selection
- refresh() is called when visible - Telemetry page with script displayed for BW or Main View with Widget script is visible
- background() is called when not visible - visible means also widget is obscured by OS UI element
- both refresh() & background() is NOT called if One-Time script (1) is running
script is stopped and disabled if it misbehaves (e.g. too long runtime, run-time error, or low memory)

3. Custom scripts (Mixer) 
- init() is called at script start
- run() is called all time.
- run() NOT called if One-Time script (1) is running
- script is stopped and disabled if it misbehaves (e.g. too long runtime, run-time error, or low memory)

4. Function scripts
- init() is called at model selection  
- run() is called when function switch is off
- background() is called when function switch is off
- script is stopped and disabled if it misbehaves (e.g. too long runtime, run-time error, or low memory)

Now testing  in 2.8
https://youtu.be/-65lFoFJk1I

Test widget
[BKGTEST.zip](https://github.com/EdgeTX/edgetx/files/10011527/BKGTEST.zip)

lshems, 2022-11-15:
Please define temetru scripts apart from widgets. They are different things.

Background for telemetry scripts get always called, 'run' only when on screen (even of the annoying telemztry reset is shown).

I know it is to much asked but you could consider reducing the number of script formats by joining the one time and telemetry scripts. I already do so by always calling the background from the run and returning a 0, so a telemetry script can be run as a one time as well.

This would also imply to return to the old setup to NOT call the background on a telemetry script would be logical. This to prevent the background from being run twice for such a universal script where you need the background to run in 'run' view as well.

pfeerick, 2024-06-17:
Resolved by #4393

closed 2024-06-17 by pfeerick