EdgeTX/edgetx #2715: Since 2.8 I sometimes crash my TX16S with a message "emergency mode"

Billy60, 2022-11-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Randomly but usually whils programming for a long time, I get the black screen of death with "emergency mode" .
I then have to start the Tx again.
What if this happens in flight ?

### Expected Behavior

No crash !

### Steps To Reproduce

use the Tx for a longish time : about 1/2 hour
randomly get the crash + error message

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Edge 2.8RC4

pfeerick, 2022-11-12:
More information is needed if you want any hope of this fixed. We *need* to have some reproducible cause/trigger for the crash in order to start fault-finding. It's unlikely to be random, just a matter a find out that sequence of steps needed. 

As far as "What if this happens in flight?" - depending on the protocol and exact setup, the radio UI is halted, but the RF and basic controls are still functional - this is the purpose of Emergency Mode.

gagarinlg, 2023-09-20:
it might be worth retesting this now, with the new SD-Drivers merged into main