EdgeTX/edgetx #2719: No Sound

oldnansome, 2022-11-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Have beeps but no voice. Voice files show in firmware.

### Expected Behavior

Voice

### Steps To Reproduce

Openin splash screen expect to hear welcome and also when switches moved

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-11-13:
- Which audio pack?
- Do you have Radio Settings -> Radio Setup -> Voice language set correctly?
- Are the files correctly located in /SOUNDS/## , where ## is the two letter language specifier? (e.g. the audio for "3D" for english is `/SOUNDS/en/3d.wav`)
- Do the files play ok if you play them via the Radio Settings -> SD browser? (highlight one, tap/press enter, and choose "Play")
- If you expect it to play sounds when switches are moved, are there special functions configured for this, or are you using the model dependent folder triggers?

oldnansome, 2022-11-13:
They play and yes I expect them to play when setup model. Does that help isolatethe issue?

closed 2022-11-14 by pfeerick