EdgeTX/edgetx #2720: Firmware Switches S1 and S2

oldnansome, 2022-11-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I cannot seen the switches S1 and S2 on the switch drop down menu. They do however show up on the model screen and register when being moved

### Expected Behavior

S1 and S2 to be visible options on drop down menu when assigning switches

### Steps To Reproduce

Turn on the radio and go to menu requiring switch asignment

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

eshifri, 2022-11-12:
S1 and S2 are not switches - they are potentiometers.

oldnansome, 2022-11-12:
Thank you, so were do i find them on the switch list cos there aint no potentiometer switch  or am i missing an icon

eshifri, 2022-11-12:
Can you please provide more details regarding what you are doing? Step by step.
S1 and S2 are not in the switches list, since they are not switches.
They are available as sources.

oldnansome, 2022-11-13:
Assigning global values, the option to use S1or S2 doesn't come up.

pfeerick, 2022-11-13:
Do you mean the "Adjust" special function - as in Adjust -> GV# -> Mixer Source ?

As S1 and S2 do appear in that list. However, you cannot choose them in the first dropdown, as S1 and S2 are potentiometers, not switches. The first option, "switch", controls when the special function is actually active - you can set it to ON if you want it active all the time. 

e.g.

https://user-images.githubusercontent.com/5500713/201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4

oldnansome, 2022-11-13:
I am trying to use them to adjust the volume

https://user-images.githubusercontent.com/5500713/201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5500713%2F201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4&data=05%7C01%7C%7Cf03dcd6fbf4443648ec708dac574ddb7%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638039402975279455%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=mt5azOlU60oULRUvyAe6DIalRs6KC%2FWy3hIZegjGoH8%3D&reserved=0>

oldnansome, 2022-11-13:
I had no sound though the en library folder is shown in  the sounds folder. I renamed the fo

https://user-images.githubusercontent.com/5500713/201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5500713%2F201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4&data=05%7C01%7C%7Cf03dcd6fbf4443648ec708dac574ddb7%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638039402975279455%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=mt5azOlU60oULRUvyAe6DIalRs6KC%2FWy3hIZegjGoH8%3D&reserved=0>

oldnansome, 2022-11-13:
I had no sound though the en library folder is shown in the sounds folder. I renamed the folder en and now I have sounds/tracks that play when assigned to switches etc. Seems the software was unable to find the folder it was looking for. Now renamed it works. Hope that's just a hiccup but surprised no one else has had that problem.

https://user-images.githubusercontent.com/5500713/201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5500713%2F201522293-3a636cba-3f79-4185-a31e-eac14d3b5b86.mp4&data=05%7C01%7C%7Cf03dcd6fbf4443648ec708dac574ddb7%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638039402975279455%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=mt5azOlU60oULRUvyAe6DIalRs6KC%2FWy3hIZegjGoH8%3D&reserved=0>

pfeerick, 2022-11-13:
What was the folder called before, and what did you rename it to?

pfeerick, 2022-11-13:
> I am trying to use them to adjust the volume 

Is there a specific reason you're trying to do this via a GV rather than using the dedicated volume special function? 

https://user-images.githubusercontent.com/5500713/201550574-fd9ee2d1-585f-47e9-8bb9-24dd3cce597b.mp4

oldnansome, 2022-11-14:
The folder, downloaded was "en-gb-libby" on looking at sound folder prior to updating the folder was called simply "en". Changed folder to that name and i now works, including the welcome sound which previously failed to work. I am using the GV system having looked at "rc radio revie" youtube channel. I am unaware of the model dependant trigger. All of the open/edge tx is new to me so i am hopefully learning as i go along. I am just grateful that people like your good self are available to help.
It is presently not good weather to fly but I am hoping to have one of my planes set up on this system and thengo and test fly the transmitter!
Regards
Mick

pfeerick, 2022-11-14:
Ok, lovely. That is a known issue / gotcha with the variants of the different language packs. i.e. for English, all but Irish Emily will need to have the downloaded folder renamed to `en` in order for the radio firmware to see it. 

John does some great explainer videos, and also has a [discord group](https://discord.gg/jSFTASU) where you can ask questions (and you can ask questions and get help on [our own discord group](https://discord.gg/wF9wUKnZ6H) as well, we don't bite! ;) ) 

Using special functions to play sounds on movement of switches is pretty much the normal method of doing it. There is another method, which relies on the naming of the files, but that is not as flexible as it does require you to use the computer to make any changes, rather than being able to do it on the radio itself, as well as needing to have multiple copies of the audio files, but it is probably useful for some people. 

Good luck! Hopefully, you get everything working as you want it without too much hair pulling ;)

closed 2022-11-14 by pfeerick