EdgeTX/edgetx #2721: NV-14 Phanton Telemetry RPM from FC's

Outwide, 2022-11-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Phantom RPM telemetry showing in NV-14 display telemetry display. https://github.com/EdgeTX/edgetx/labels/bug%20%F0%9F%AA%B2
Telemetry display when using BF4.3 Fr 10.8 shows one RPM set at 1000
Telemetry display when using Inav 5.1 shows two RPM set at 0 and the other counting up

[image]
[image]

### Expected Behavior

Only show actual telemetry transmitted, No phantoms. https://github.com/EdgeTX/edgetx/labels/bug%20%F0%9F%AA%B2

### Steps To Reproduce

I have setup test rigs for both HGLRC Zeus F722 and Matek H743 Slim both FC's can be used with either BF 4.3 Fr10.8 and Inav 5.1.

Clean install of BF 4.3 Fr10.8 both FC's (HGLRC Zeus F722 and Matek H743 Slim) shows one phantom telemetry RPM which is wrong. No motor or rpm sensor is attached.

Clean install of Inav5.1 both FC's (HGLRC Zeus F722 and Matek H743 Slim) show two phantom telemetry RPM which is wrong. No motor or rpm sensor is attached.

With Matek H743 Slim I tried it with motor and without but go the same result.

### Version

Other (Please specify below)

### Transmitter

Flysky NV14

### Anything else?

I like Edge TX you guys are doing fantastic work.
2.8.0 RC4

Outwide, 2022-11-18:
> Can you please provide some additional info/try some experiments:
>
>   1.  Which receiver are you using?
>   2.  I assume you sending FC telemetry to the iBUS input of the receiver, correct?
>   3.  If you remove all sensors and run rediscover - doe the problem persist?
>   4.  A3 telemetry - is it coming from the FC or from the FlySky sensor?
>
> ETX for NV14 has a very limited number of predefined sensor. RPNM is one of them.
> If the telemetry signal is interpreted as RPM sensor, it means that the FC puts a corresponding ID (number) into the packet.
> RX and FC do not talk with other to agree on the sensor IDs.
>
> As a work around you can jut remove phantom sensors from the list, if this is an acceptable solution.

1.  Which receiver are you using

  *   FS-IA6B
  *   FS_IA10B
  *   Note: I will also check with the, Crossover receiver XR602T-A and ELRS which is coming soon and let you know what I found.
2- I assume you sending FC telemetry to the iBUS input of the receiver, correct?

  *   Yes from the FC to the receiver and all working. You can see the wires at the bottom that connect to the receiver and I-Bus sensor. Picture attached

  1.  If you remove all sensors and run rediscover - doe the problem persist?

  *   Yes I have done rediscover about 10 times and have been trying to find this problem for months.

  1.  A3 telemetry - is it coming from the FC or from the FlySky sensor?

  *   The FC . No fly sensor or esc is attached. The A3 appears to be working fine and is the battery voltage no need to investigate this for me.

Just so you know I have been trying to get and RPM sensor working from the FC for months but it has appeared that I have been getting Phantom telemetry. The picture above is as you see it, clean install of operating programme with no ESC’s or anything else attached.

Note: The Flysky sensors work and I think with no problem. These are value sensor type.

I wish to use all sensors from the FC, which are a telemetry type sensors,  and add other telemetry to the FC and then out to EdgeTX.

Thank you for your reply and if I need be clearer let me know.

I like Edge TX

eshifri, 2022-11-18:
It may be completely irrelevant, but on the second image there are two A3 sensors showing different voltage.
Is it correct behavior?

I have never tried any RPM sensor, so there is a chance RPM packets decoding never worked for NV14.

Outwide, 2022-11-18:
Yes I am aware of A3, I did change the voltage scale and didn’t reset. Completely irrelevant.

On the picture I supplied of the 2 FC no esc or telemetry sensor is attached. I get the same phantom RPM telemetry on the 2 other Matek H743 slim FC’s which are connected to esc and other devices.

I have supplied a New picture of the test rig BF4.3 10.8 and just rechecked it again and still get phantom RPM telemetry and supposably run at a 1000RPM.

I will test the other receiver soon, Crossover receiver XR602T-A and then on the ELRS when it arrives.

Note: it maybe a receiver base problem

Thanks

Outwide, 2022-11-18:
I went back to check Inav 5.1 and made sure things were turned off as per program. See pictures

Thanks

eshifri, 2022-11-18:
Sorry, I do not see any new pictures in this thread - only two original pictures.

Outwide, 2022-11-18:
Sorry they just didn’t send

pfeerick, 2022-11-19:
Images and files attached to emails are rejected by github - you'll need to upload them on the issue page (i.e. drag and drop the images to the text entry area to upload).

Outwide, 2022-11-19:
Sorry, these are the images as per discussion above.

[image]
[image]
[image]
[image]

eshifri, 2022-11-19:
Looks like you have old internal radio module FW (Sgnl==10 on your images).
I have no idea whether it will help, but you may want to try an updated version from here:
https://www.rcgroups.com/forums/showpost.php?p=45357535
Some discussions may be found here: https://www.rcgroups.com/forums/showthread.php?3023832-UndergroundFPV-Dark-Knight-Radio/page251

Telemetry processing is handled differently with the new FW (more or less on the MPM route).

Outwide, 2022-11-19:
Hi thanks for the info. I ask how do I flash the "internal chip" FW and keep all the Edge TX info and working. I saw something about a special loader for that prepose, said it was under development.  
I just want to be very careful ! as I am not sure.

[image]

eshifri, 2022-11-19:
There is an unpleasant surprise: this screen shows you have the latest version.
Unfortunately, there are two different radio module FWs in the field with the same version number.

In order to update the radio FW please be advised to put it into the FIRMWARE folder on the SD card (unzip it first, of course).
On the TX navigate to this folder and select the file. In the pop-up menu there will be an option to update the radio FW.

There is also one more thing we can try, maybe before that. I've got some ideas, but I cannot test them. Do you think you will be able to give a try for the ETX version I upload?

Outwide, 2022-11-19:
Yes

Outwide, 2022-11-19:
"Yes" Please I would like to try whatever method you recommend.

eshifri, 2022-11-19:
[firmware.zip](https://github.com/EdgeTX/edgetx/files/10047933/firmware.zip)
Can you please give it a try?

Outwide, 2022-11-19:
Ok I have put the new firmware in to the FIRMWARE BIN on the SD card and I will open the TX companion load it as stated above.

Outwide, 2022-11-19:
Problem,  The NV-14 connected to the computer, using USB, see on screen on the Transmitter. In TX Companion I navigate to the SD card on the handset and FIRMWARE BIN and load the FIRMWARE file you sent. This was my procedure.  
[image]

eshifri, 2022-11-19:
This is just an ETX FW. The fastest way to install it is from SD card: just copy it into the FIRMWARE folder, boot the TX in the bootloader mode (press both trims towards the center and switch the TX on). You will be able to install a new FW.

eshifri, 2022-11-19:
Actually, I'm not sure Companion can install the FW that is on the SD card - never tried this.

eshifri, 2022-11-19:
And one more version to try.
[firmware.zip](https://github.com/EdgeTX/edgetx/files/10047958/firmware.zip)

Outwide, 2022-11-19:
Ok If I go into Bootloader mode I am not able to see the SD card.
I have tried moving both Blue button trims toward the centre of the screen and setting to Bootloader mode but I can't seed the SD card on the computer or tell TX to go get it from the bin.

Can I just tell TX to go to my computer and load the firmware.

eshifri, 2022-11-19:
There are many ways to do it, but the fastest is:
1. Switch TX on, Connect it to the computer with USB, copy firmware.bin into the FIRMWARE folder.
2. Disconnect the USB and switch the TX off.
3. Switch TX on in the bootloader mode. There will be and option to "Write the firmware". 
4. All FWs on the SD card will be listed.
5. Follow the on-screen instructions.

Please start with the latest  version I have uploaded.

Outwide, 2022-11-19:
3. Switch TX on in the bootloader mode. There will be and option to "Write the firmware".
When I do this it is jut a black screen. Not comes up you can only to your computer

OK I can start the Transmitter and navigate to the Radio setting Tab and to Firmware bin and load the select the firmware and flash.
 
[image]

Outwide, 2022-11-19:
[image]

Outwide, 2022-11-19:
Ok I have deleted the old telemetry screen and new discover. 
[image]

eshifri, 2022-11-19:
Unfortunately, this is flashing a bootloader only.
To flash the FW itself you need to be in the bootloader mode.

Let's check you are doing it correctly: press both trims towards the center and both power buttons simultaneously.
You should see the bootloader menu.

OK, this is from my latest build?  
I do not know what to make from this: there are two RPM sensors in the FlySky system. So it might be possible that one returns zero and the other - random number.

Outwide, 2022-11-19:
Yes it worked this time. RPMs are a zero both.
[image]

eshifri, 2022-11-19:
Aha! Some progress! Thank you for testing this!
Do you have any way to test it with the motor? To see measured values?

Outwide, 2022-11-19:
I would have to say No because as Inav uses the belhli32 to supply RPM signal, for the smaller ESC, to the OSD only not out to the Transmitter. I have asked for it to be considered on the Inav github but as I am a no one so it may not be heard. I am wanting to get a 2 wire system such as the hobby wing RPM sensor, which from my investigation, only outputs a signal of difference which then has to be coded in the FC so it can come out to the OSD and Transmitter. This what I am ask help for, someone to do some coding, as it is very important for preflight checks and documentation for drone licensing, as if there is something wrong you will see it progress.
Note: I have contacted CoderElectronics to see if he can help me with some code you may wish to touch base with him. All help is good.

Note: I only have the original Flysky RPM sensor which I had no problem with but not suitable with FC

Outwide, 2022-11-19:
I personal say Thank you for your generous help. You are truly helping me and other so much.

eshifri, 2022-11-19:
You are most welcome. 

Are you saying FlySky RPM sensor worked properly without my changes?

Outwide, 2022-11-19:
I can't say that yet as I don't have a rig setup for that yet. My understanding is that FlySky RPM sensor is a value sensor and FC's transmit "telemetry". Tricky wording which mean you can't connect them both together as one cancels out the other.

I am tiring to get all telemetry out of an FC and I am nearly there. Just the 2 wire system such as the hobby wing RPM sensor left.

Outwide, 2022-11-19:
I would say that the issue can be closed for now as both RPM's show zero.

eshifri, 2022-11-19:
Again, I have never tried, but all FlySky sensors are digital and communicate with RX through iBUS (Serial port). They can be daisy-chained. You can have several sensors reporting the same value - they will be assigned different IDs and will be visible.

Outwide, 2022-11-19:
Yes I have tried that. But if you are wanting a small footprint in your aircraft an FC is a good way to go as you get a lot of telemetry output.

Outwide, 2022-11-19:
Hi I ask would it be ok to keep in contact with you. I am on Discord it would be nice to have some friends I can talk with.

eshifri, 2022-11-19:
Sure!

eshifri, 2022-11-19:
Actually, it should not be closed since the code was not committed yet.

pfeerick, 2022-11-20:
Reopened

> Actually, I'm not sure Companion can install the FW that is on the SD card - never tried this.

Probably not - since you really should be in DFU mode when doing that (so the SD card will not be visible). Online flashing when the SD card *is* visible is just begging for disaster. Thus flashing via the bootloader on the transmitter itself is the simplest, safest and fastest... 

Just for reference, with 2.8.0-RC4 firmware, I get the following when the NV14 and FS-iA6B have no sensors connected, 

[image]

If I have either the CPD01 (Magnetic) or CPD02 (Optical) Sensors I get only one extra sensor entry:

[image]

If I have both connected, I get two entries, with different instances:

[image]

Finally both the CPDO1 and CPD02 working... they pretty much agree on the speed of the motor, which was nice. 

[image]

eshifri, 2022-11-20:
Hm, then it should be probably closed "with no problem found".
I'm a little bit surprised it works as is, since I thought I have found a problem. :-)
If it works with native sensors the problem must be on the FC side, right? 
The only difference I see here is you are using the updated radio module FW (Sgnl==99).
The telemetry processing is different for old and new versions:
https://github.com/EdgeTX/edgetx/blob/f84c26b6152b3cd1a497bea49eca4fbc95b5e1c1/radio/src/telemetry/flysky_nv14.cpp#L272
My change, most likely, affected the older FW route.
Unfortunately, I do not have these sensors and cannot test.
Shall I submit a PR? You will be able to see whether it breaks something. :-)

pfeerick, 2022-11-20:
Sure, can at least give it a try and let you know. :) 

Yes, I updated the NV14 internal RF back when you first told me there was a newer FW for it... and that really does need to be the minimum standard here since it exists and fixes things. So if it only affects older FW, users of that are on their own unless they complain 🤷‍♂️ . 

Until a comparison is done to show telemetry with the FC connected, and telemetry without the FC connected (after sensor deletion and rediscovery, of course) which proves otherwise, I would strongly be leaning towards something in the Betaflight/iNav telemetry implementation for iBus is a bit broken, thus that time tested IT adage of "💩 in, 💩 out" would be applicable here.

Outwide, 2022-11-20:
Hi, again thank you both so much for your help. So I ask, what should I do? get the new Main board V1.6. Will you guys let the Dev's in Inav 5.1 know there is an issue in their software as a common curtsey. 
Just so you know I still have to test another two receivers connect to an FC to see what I get. and then I can reconnect back to the old style "Box receiver "CPD02 (RPM), Temp and Voltage.

eshifri, 2022-11-20:
I do not think  you should change the main board unless you have other reasons to do it.
You MAY want to upgrade the radio module FW (my links above).
I have submitted a PR and would suggest you wait for some testing.
Since you do have the FlySky sensors you may want to check how they work with ETX before and after the fix.
Do not rush. :-)

Outwide, 2022-11-20:
I see it has finished checking do I need to get this file and how or just wait.

eshifri, 2022-11-20:
It is exactly the same binary file I gave you above.

pfeerick, 2022-11-20:
Specifically, as mentioned in this comment https://github.com/EdgeTX/edgetx/issues/2721#issuecomment-1320950171. You copy it onto your radio, put it in a folder (I suggest FIRMWARE also for this), and then with the radio running normaly, go into radio settings, SD, FIRMWARE and select the file - you'll get an option to flash the internal rf module.

Outwide, 2022-11-23:
Hi eshifri how do I contact you as I couldn't see you on Discord.
I have tried this receiver, Crossover receiver XR602T-A which has telemetry on the HGLRC Zeus F722, using Inav 5.1 and was not able to get it to work in Inav, was able to get it to bind. I also tried in with Beta flight 4.3 and was able to get it to work in Beta flight and see stick movement in the receiver section but kept getting RF signal warning critical. I have noticed at the top right-hand side, signal bars where all black. I will try with my other TX tomorrow which doesn't have the update on yet and if there is a change, I will let you know. ExpressLRS does not appear to be ready yet form the NV-14. It would appear that there are few receivers with telemetry for Flysky AFHDS 2A and AFHDS 3 is still broken.

eshifri, 2022-11-24:
Hi @Outwide I'm at the Discord under the same name. You can find me in ETX forum (usually, in the dev's section).
I think ELRS works on NV14, but again: I have never tried. :-)
RF signal warning may be misleading - it is not standardized over different protocols and it is possible the warning levels are incorrect for your protocol. Please take a look at the signal quality numbers in the telemetry (if you receive it).

pfeerick, 2022-11-24:
ELRS/CRSF are both working fine for me with the NV14 in both 2.7.1 and 2.8.0. 

I don't know anything about Crossover XR602T-A other than it exists, and seems to use AFHDS2A, but the Flysky iA6B and iA10B and sensors are working fine as well. I had to update the firmware on the Flysky iA6B though (that was fun!) as I found it had a problem with channel outputs with the original firmware it shipped with (was a really old version). That's the only major issue I've had (other than the NV14 internal RF needing to be updated also to fix bugs like the signal quality value).