EdgeTX/edgetx #2722: Selecting a model that has "Display checklist" turned on...

Astrohog, 2022-11-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

TX16s MKI running 2.8 RC4
Enter model setup for model.  Turn on "display checklist" on model setup tab.  Select different model.  Go back and select first model (one that should display the checklist).  After selecting the model I may get a throttle and or switch warning.  After clearing the warnings I'm back at the model select page rather than the checklist.  If I press the return button, I exit the model select with the correct model loaded but no check list is displayed.  The checklist is displayed when the power is cycled on the radio.  In short the model checklist is only displayed when the power is cycled and not when switching from one model to another ( one with Display checklist turned on).

### Expected Behavior

I expect that when selecting a model that has the option "Display checklist" turned on that the after selecting the model the checklist is displayed (may have to clear warnings first).  I can then press the return button to exit the checklist and the selected model is loaded. 

### Steps To Reproduce

Load Model with "display checklist" turned on.  Switch to another model.  Select the first model.  Verify that the checklist is displayed.  May have to clear warnings first.  

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Firmware 2.8 RC4

pfeerick, 2022-11-20:
I *think* the underlying issue is because at the time this code runs, the code here checking for model notes doesn't know about the new model name... 
https://github.com/EdgeTX/edgetx/blob/f84c26b6152b3cd1a497bea49eca4fbc95b5e1c1/radio/src/opentx.cpp#L741-L743

But there is also a secondary issue, which is that pulses is running when the model notes are shown, rather than halted until the checklist is closed.

eshifri, 2022-11-28:
Actually, the model name is set there.
The problem is for some reason the ViewTextWindow() is not visible, when called from model_select.
The code is executed, but the text is hidden.
When the model with the notes is selected, we do not return to the main screen, but stay on the selection screen. The RTN is intercepted by invisible text viewer.

PR #2922 (2022-12-22): fix(color): Display of model notes when selecting model
Alternate solution to the one in PR #2792

My testing on TX16S and in the simulator has no issues; but if there are reasons why doing it this way won't work please let me know.

Fixes #2722 and #2921

Summary of changes:

Close model select page before loading model so notes window is not hidden.
closed 2022-12-22 by pfeerick