EdgeTX/edgetx #2726: Trainer mode cable connection intermittent

FPVxfun, 2022-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While connecting the 2 transmitter (master and slave) with cable, connction is often lost. Alle the parameter of connection (timing, polarity) are at default (CH1..CH8, 22.5 ms, 300 us, negative polarity).

I'm sure it is not an issue of the cable because I normally use that cable for PC simulator. In any case we have tried another cable and the behaviour is the same.

I've tested the same issue between 2 Radiomaster TX16S and between 1 Radiomaster TX16S and 1 Taranis X9D that worked perfectly in trainer mode with other radios.

### Expected Behavior

Trainer connection signal improvement

### Steps To Reproduce

1. Connect the 2 radios with trainer cable, 
2. Turn-On both radios in trainer mode (one master, other slave), default PPM parameters
3. Show output channer in the main display
4. Listen frequently to frequently "connection lost" and "connection recovered" radio messages and observe output channe freeze on the display

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-11-27:
Can you try again with 2.8.0 which has just been released, and see if the issue is still present. 

And can I confirm - you are saying the issue is only between the two TX16S, and there is no issue when it's TX16S -> X9D?  (All running ETX 2.7.1). Or there is an issue between all of them, but no issue when using them in conjunction with some other radio?

FPVxfun, 2022-11-27:
It happens in any case with Radiomaster TX16S. I have tried both between my 2 TX16S and between one of my TX16S and X9D of my friend. X9D of my friend (and my friend's cable) is working for sure since my friend uses it to teach flying to other people.

dlktdr, 2022-11-27:
The original TX16s has a non-standard trainer socket it's a TRRS that is 16.8mm long. Without an adapter or modification to the port a standard cable won't work. 

Here is the available adapter,

How to modify the port

The TX16S Mark 2 doesn't have this issue, it uses a standard plug.

FPVxfun, 2022-12-03:
Dear dlktdr I have followed all the instruction on rcgroups and now master-slavw works great. On the next few days I will install EdgeTX 2.8. many thanks for your support

closed 2023-05-05 by gagarinlg