EdgeTX/edgetx #2727: AdjustRTC CHeckbox missing in Radiofirmware/Simulator

ParkerEde, 2022-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I miss in the radio firmware, as well as in the simulator in the radio basic settings, the checkbox to set the time automatically and the field "timezone". I think it is about the flag "adjustRTC" and "timezone" in radio.yml. In Companion, however, it is displayed:

[image]

In Firmware I miss this
[image]

### Expected Behavior

"adjustRTC" and "timezone" should be configurable in Radio

### Steps To Reproduce

1. open Radio Settings
2. try to adjust timezone and adjustRTC

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild

pfeerick, 2022-11-13:
Not missing... just moved ;) Check under the GPS button, since these are GPS options ;)

[image]

ParkerEde, 2022-11-14:
@pfeerick Thank you for your response. I don't have the GPS button on my HorusX10 Express. What do I have to do to get the GPS button?

pfeerick, 2022-11-14:
Hm... probably need to enable internal GPS support then as it's probably not enabled at all for that target... I'll check and get back to you soon...

pfeerick, 2022-11-14:
Ok, yes, so it's dependent on `INTERNAL_GPS` being defined. And it seems like it's not valid for that target... so the next question is "under what circumstances is it valid for the radio to show those options?" ... as I suspect it's not, and it's actually Companion that shouldn't be showing them?

ParkerEde, 2022-11-14:
This option must always be displayed on all transmitters, because it uses the time only from the GPS telemetry sensor. The internal GPS module has nothing to do with the time determination.

pfeerick, 2022-11-14:
Well, ouch... so it does... and the code even says it comes from telemetry, not internal GPS 🤦 I'm just glad it wasn't me that changed that code 🤣 Yes, this needs to be re-opened, and at least re-enabled until implemented differently.

ParkerEde, 2022-11-14:
Can we still get that into the 2.8?

PR #2732 (2022-11-16): fix: Always show GPS button in radio setup
It is related to GPS telemetry, not internal GPS.
This probably needs to implemented differently in the future.

Fixes #2727

Summary of changes:
closed 2022-11-16 by pfeerick