EdgeTX/edgetx #2728: Logical switch changed from RPM to TMP when new sensors are discovered

Kevltan, 2022-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I remove and re-discover new telemetry sensors my RPM sensor in my active model is changed from RPM to TMP in logical switches. Screen theme turns red including shut down screen. On restarting everything appears to be OK except the RPM is changed to TMP in my L02 logical switch. Works fine when edited back to RPM. Question is why does a telemetry "discover new" edit my model data?

### Expected Behavior

Nochange in theme.
No change to model data.

### Steps To Reproduce

1. Delete sensors
2. Re-discover new sensors
3. Red theme
4. Red shutdown theme
5. Restart
6. Original theme returns
7. Changed RPM to TMP in logical switch noted.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

2.8.0 RC4 firmware.

eshifri, 2022-11-14:
Unfortunately, this is not  a bug - it is a feature.
Sensors are addressed by the index - not by name.
It is deeply rooted in OTX history.
Every time you run re-discovery indexes change and so...

pfeerick, 2022-11-14:
As @eshifri said, this is a long-standing issue that has its origins in OTX. Hopefully, we can finally put that to rest in the near future ;)

But in the interim, the answer would be the same as if you raised this there... don't do sensor discovery again after setting everything up, or you'll need to go through them all. As they don't necessarily get added in the same order each time you do discovery - since they get added in the order they are seen.