EdgeTX/edgetx #2733: Editing THEMES triggers "emergency mode" when closing

paul-goelz, 2022-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Editing any theme will trigger "emergency mode" when backing out after any edit.  This happens 100% of the time.  Turning the radio off and then back on again results in a clean power on and correct operation.  The edits to the theme are saved and display correctly.  

### Expected Behavior

No emergency mode......

### Steps To Reproduce

1.  Press SYSTEM > PAGE > PAGE > PAGE and select a theme.  
2.  Edit the color of any feature in a theme. 
3.  Press RTN. 
4.  Radio enters "emergency mode".  
5. Power off and then on again.  
6. Radio powers up and operates normally.  The edits to the theme are active.  

### Version

2.7.1

### Transmitter

Jumper T16

pfeerick, 2022-11-15:
This has most likely been fixed in 2.8, as there were several EM triggers related to the theme editor fixed. Can you try again with 2.8.0-RC4?

paul-goelz, 2022-11-15:
I'd be happy to.  But I have to ask first.... can I flash back and forth between 2.8.* and 2.7.* freely without affecting my models?

pfeerick, 2022-11-15:
You'll need to backup your MODELS and Radio folders first (well, actually really only MODELS but it is good practise to do both when doing any update/testing), as the change from categories to labels isn't backwards compatible. If you can get the simulator to crash the same way though, you could also try with Companion simulator instead.

paul-goelz, 2022-11-15:
Thanks!

Could I get you to contact me off this forum (or maybe on it?).  I have an unrelated question about how the external MPM signals the OS that it is in bind mode that is probably not appropriate for this forum.  I didn't see an obvious way to PM you....

gagarinlg, 2022-11-15:
> Thanks!
> 
> Could I get you to contact me off this forum (or maybe on it?). I have an unrelated question about how the external MPM signals the OS that it is in bind mode that is probably not appropriate for this forum. I didn't see an obvious way to PM you....

Join our discord server and ask there:
https://discord.gg/PQch74zD

pfeerick, 2022-11-20:
Possibly, just possibly there is an SD card issue there... as the loading text should be there just before it starts reading the image previews from the SD card. All I can say at this point is that I can't get mine to crash atm... poking around in a few color elements of the theme, setting some author details, saving it... everything seems to be behaving it self... but obviously something is still lurking somewhere.

pfeerick, 2022-11-25:
@Astrohog If you get a chance, can you see if https://github.com/EdgeTX/edgetx/issues/2785 fixed this for you (will be in the next nightly). I'm going to close this issue in the hope this is all resolved now, but feel free to comment or request it re-opened if we find it's still lurking in 2.8.0.

closed 2022-11-25 by pfeerick

paul-goelz, 2022-11-25:
I don't have a spare transmitter and I would rather not use my regular transmitter to test nightlies so I'll take your word for it ;)  Thanks.