EdgeTX/edgetx #2734: Radio occasionally freezes and returns to MODEL SELECT when trying to enter MODEL EDIT

paul-goelz, 2022-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Occasionally, when on the home screen and I press the MDL button to edit model parameters, the radio will appear to freeze and then after 5-10 seconds will display the list of models as if it was in the model select screen.  Press RTN and it returns to the home screen.  The next press of MDL always enters model edit correctly.  I have not been able to find any obvious trigger and while it happens frequently, it does not happen every time.  

### Expected Behavior

Able to enter model edit from the home screen every time MDL is pressed.

### Steps To Reproduce

See description.  I have not found a way to reliably repeat it.  But it is not a rare occurrence.  I would say it happens somewhere between 10% and 30% of the time when I try to enter the model edit function from the main page.  

### Version

2.7.1

### Transmitter

Jumper T16

pfeerick, 2022-11-15:
Sounds like you are holding the MDL key too long. For 2.7 onwards, a long press of MDL (I can't remember the exact duration, probably > 500ms) will open the model select screen. The freeze is because it's having to read all the model images, which has also changed in 2.8 with the label system - it now reads the images one screen at a time as needed.

paul-goelz, 2022-11-15:
Oh.  Did not know that.  While undesirable, at least now I know what is happening.  

Might I suggest that this behavior get changed so it does not confuse others like me?  Or is it related to touch compatibility?  Still though..... not very clean design.......

paul-goelz, 2022-11-15:
Lets call this closed.  It is very likely that I was holding the MDL button too long.

closed 2022-11-15 by paul-goelz