EdgeTX/edgetx #2735: Instructions to install EdgeTX over OpenTX are incomplete.

paul-goelz, 2022-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Not sure if this is the correct place to report this.....

The published instructions for converting to EdgeTX from OpenTX are incomplete.  The procedure as written seems to apply to a clean install but it skips one important step if you are installing OVER OpenTX.  

If I prepare an EdgeTX SDCard per the instructions and place it in a transmitter that is currently running OpenTX, the transmitter will not boot correctly and I then cannot navigate to the SDCard to install the EdgeTX bootloader.  

If I place the CURRENTLY INSTALLED copy of OpenTX firmware in the FIRMWARE folder on the SDCard IN ADDITION TO EdgeTX, the radio will boot normally and I can install the EdgeTX bootloader normally.  

### Expected Behavior

The instructions need to be re-written to take into account the situation where a user is installing EdgeTX on a radio that is currently running OpenTX.  Or have I missed something??

### Steps To Reproduce

1.  Prepare SDCard per published instructions.  
2. Place that SDCard in a transmitter that has OpenTX installed.  
3. Transmitter will not boot because the currently installed version of OpenTX is not on the SDCard.
4. Since the transmitter will not boot, you cannot navigate to the SDCard and install the EdgeTX bootloader.  

### Version

2.7.1

### Transmitter

Jumper T16

### Anything else?

I realize this is not actually a firmware bug.  If this is not the correct place to report it, my apologies.  But it is a serious issue that will affect anyone trying to move from OpenTX to EdgeTX.

pfeerick, 2022-11-14:
Do you mean this?

https://github.com/EdgeTX/edgetx.github.io/wiki/EdgeTX-Installation-Guide

If so, yes, looking at this afresh this is a bit back-to-front - You need to at least flash the bootloader before doing the rest of the SD card prep.

paul-goelz, 2022-11-15:
Yes, those are the instructions in question.  They simply do not work as written if you are transitioning from OpenTX to Edge.  They WILL work if you put a copy of the currently installed OpenTX on the Edge SDCard because you can get it to boot into OpenTX, then install the bootloader, then install Edge.  

There is also an issue where sometimes the models you copied over from OpenTX in the MODELS and RADIO folders are not recognized by Edge.  I had to use Edge Companion to open my OpenTX models file and then write them to the Edge radio.

pfeerick, 2022-11-15:
Wiki has now been updated. Thanks for bringing this to our attention! :)

Yes, it seems like sometimes it doesn't pick them up the first time. It seems more prevelant with older models that have been brought along for many versions, and can sometimes be resolved by simply deleting the converted files, and getting the radio to re-convert/re-import them. Otherwise, Companion should be able to read them all and write them back properly. Once you're onto the Yaml file format (EdgeTX 2.6 onwards) this will be a thing of the past.

closed 2022-11-15 by pfeerick