EdgeTX/edgetx #2736: LUA key events missing in 2.8 RC4 for ENTER & EXIT key (regression)

JimB40, 2022-11-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On 2.8 RC4 color tagets (TX16S, X10, X12 etc)

ENTER key generates:
2050 | EVT_ENTER_LONG
514 | EVT_ENTER_BREAK

EXIT key generates (only)
516 | EVT_EXIT_BREAK

### Expected Behavior

Events that should be generated (like in 2.7)

ENTER key
1538 | EVT_ENTER_FIRST
2050 | EVT_ENTER_LONG
1026 | EVT_ENTER_REPT
514 | EVT_ENTER_BREAK

EXIT key
1540 | EVT_EXIT_FIRST
2052 | EVT_EXIT_LONG
516 | EVT_EXIT_BREAK

### Steps To Reproduce

check recieved LUA events

### Version

Other (Please specify below)

### Transmitter

FrSky X10 / X10S (ACCST), FrSky X12, Jumper T16, Jumper T18, Radiomaster TX16S / TX16SMK2

lshems, 2022-11-14:
#2664

JimB40, 2022-11-15:
KEY_EXIT_LONG won't be passed to LUA as it is used by OS event handler to terminate LUA.
However missing 
1538 |  EVT_ENTER_FIRST
1026 |  EVT_ENTER_REPT & 
1540 | EVT_EXIT_FIRST 
is a problem with existing scripts compatibility.

JimB40, 2022-11-16:
Here is simple script to test
```
return { run =
  function(evt,ts)
    lcd.clear()
    lcd.drawText(0,0,'Key event test')
    if evt and evt > 0 then
      print('Key event received',evt)
    end
    return 0
  end
}
```
[evt_test.lua.zip](https://github.com/EdgeTX/edgetx/files/10019850/evt_test.lua.zip)

More over seems like PR #2646 hasn't fixed mentioned in #2617 problem and EVT_VIRTUAL_ENTER_LONG | 2050 still appears between EVT_TOUCH_FIRST | 9844 and EVT_TOUCH_BREAK | 8820.

@raphaelcoeffic @jfrickmann @pfeerick did you test it? or its regression?

lshems, 2022-11-16:
? Event long should be after first and before break, no ?

JimB40, 2022-11-16:
Yes. Events order are
1. For keys:
a. EVT_xxx_FIRST 
b. EVT_xxx_LONG 
c. EVT_xxx_REPT 
d. EVT_xxx_BREAK

2. For touch
a. EVT_TOUCH_FIRST (one-time - lcd touched)
b. EVT_TOUCH_TAP (one-time - optional if touch & break was quick, next touch events are suppresed)
c. EVT_TOUCH_SLIDE (one-time - optional if finger moved more than predefined value)
d. EVT_TOUCH_BREAK (one-time - lcd untouched) 

In 2.7 EVT_TOUCH_LONG  was not impemented by @jfrickmann 
But in 2.8 it seems EdgeTX with LVGL generates properly long tap event - just instead of unique touch id it sends key id: EVT_ENTER_LONG
So maybe this event id  can be swapped to new defined constant EVT_TOUCH_LONG and w will have proper touch events queue:
a. EVT_TOUCH_FIRST (one-time - lcd touched)
b. EVT_TOUCH_TAP (one-time - optional if touch & break was quick, next touch events are suppresed)
c. EVT_TOUCH_SLIDE (repetetive - optional if finger moved more than predefined value)
d. EVT_TOUCH_LONG (one time - optional if lcd touched for a longer period)
d. EVT_TOUCH_BREAK (one  time - lcd untouched)

jfrickmann, 2022-11-16:
Since @raphaelcoeffic has updated the GUI to LVGL and also undone some of my previous work, I will defer to him.

pfeerick, 2022-11-17:
> More over seems like PR #2646 hasn't fixed mentioned in #2617 problem and EVT_VIRTUAL_ENTER_LONG | 2050 still appears between EVT_TOUCH_FIRST | 9844 and EVT_TOUCH_BREAK | 8820.

No, I believe the core issue there was fixed - the touch widgets were *unusable* at that point as tap counting wasn't working properly. And as mentioned in #2646, 

> There is still a spurious EVT_VIRTUAL_ENTER_LONG that gets fired during any long press event/slide. i.e. if you press and slide and hold for a moment, you'll get:
> 
> EVT_TOUCH_FIRST
> EVT_TOUCH_SLIDE
> EVT_VIRTUAL_ENTER_LONG
> EVT_TOUCH_SLIDE
> EVT_TOUCH_BREAK

There is more work needing do be done on the underlying touch drivers before it makes any sense to tacking this further, as mentioned at https://github.com/EdgeTX/edgetx/issues/2617#issuecomment-1295941422

closed 2025-05-16 by JimB40