EdgeTX/edgetx #2739: Windows build for NV14 libsimulator fails

eshifri, 2022-11-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

When building libsimulator for NV14 build fails with the error message:
D:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/edgetx-nv14-simulator.dir/objects.a(pxx1_settings.cpp.obj):D:/msys64/home/eshif/edgetx/edgetx_main/radio/src/gui/colorlcd/pxx1_settings.cpp:49: undefined reference to `isExternalAntennaEnabled()'
D:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/edgetx-nv14-simulator.dir/objects.a(pxx1_settings.cpp.obj): in function `operator()':
D:/msys64/home/eshif/edgetx/edgetx_main/radio/src/gui/colorlcd/pxx1_settings.cpp:57: undefined reference to `checkExternalAntenna()'
collect2.exe: error: ld returned 1 exit status

File /pxx1_settings.cpp is not required for NV14 and is not included into FW build. 
For simulator build the file is included.
The problem does not exist on Linux.

It can be fixed by adding 
#if defined(EXTERNAL_ANTENNA), but, probably, fixing it in the make files would be cleaner.

### Expected Behavior

Build on Windows should work.

### Steps To Reproduce

cmake -Wno-dev -DPCB=NV14 -DDEFAULT_MODE=2 -DCMAKE_BUILD_TYPE=Release ../
make configure
make -C native -j`nproc` libsimulator

on Windows.

### Version

Other (Please specify below)

### Transmitter

Flysky NV14

### Anything else?

Current main. The problem was created about a week ago.

pfeerick, 2022-11-15:
I think there is something else going on in your build system ... clear out your build folder and run cmake again... as I can't repo this here, and the windows build of companion aren't failing either (they run on windows msys runners).

[image]

pfeerick, 2022-11-15:
btw, your cmake is missing `-G "MSYS Makefiles"`

eshifri, 2022-11-15:
Clean build indeed resolved it.

closed 2022-11-15 by eshifri