EdgeTX/edgetx #2750: TX12 MK2 with 2.8 RC4 Inputs screen does not show switches

ReinhardZ1, 2022-11-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If a custom curve is used in the INPUTS screen, the assigned switch is not shown on the display (but it works).
The switches are shown, if no curve is used (e.g. if expo is used).
I've observed this for standard input (IThr) and self defined inputs.

The bug has occured in 2.8 RC2 and 2.8 RC4 (other versions not tested)

### Expected Behavior

If a switch is defined for activation of an input, that switch should be shown on the INPUT display for all types of inputs.

### Steps To Reproduce

1.) Define curve in CURVE menu
2.) Assign that curve to an input (e.g. IThr) in the INPUT menu
3.) Assign a switch for that input
4.) Check, if the switch is shown an the INPUT screen, if you leave the edit mode

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX12 / TX12MK2

### Anything else?

[image]
[image]

pfeerick, 2022-11-20:
I'm afraid I can't reproduce this... it seems like there must be something more to this... Are you able to attach the model#.yml for an offending model (or zip the MODELS and RADIO directory and upload that to the issue (drag it to the text input area). 

i.e. this is 2.8.0-RC4 for me on TX12MKII
[image]

ReinhardZ1, 2022-11-20:
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/10049474/MODELS.zip)
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/10049475/RADIO.zip)
The switch problem occurs with model00.yml (K110). RF is via external 4in1 modul (iRangeX Plus).
There is also another strange problem with model06.yml. The timer T1 does not start, if L01 is true. Maybe you can have also a look to this model.

pfeerick, 2022-11-22:
Ok, so I see the same thing as you now. 

It's a little confusing as to what happened as you named those mixes, and gave the same name as the curve. When you specifiy a mix name, it looks like neither the curve or switch info gets displayed. So if you change the name of those throttle mixes back to being blank (`---`) it will show the curve and switch info then. 

For the timer, it's because of Timer reset SF you've set is triggered when SWD is up, not when it's pressed, so is always running ;)

ReinhardZ1, 2022-11-22:
Thank you very much for your help Peter!
So the solution would be to show only 3 characters of the name for smaller screens, but that is a minor issue.
The effect with the timer reset is funny. It's easy to define the wrong switch direction for the push buttons by only pressing them ...
I think this issue can be closed now.

pfeerick, 2022-11-22:
That could be a way, but then possibly also leaves the question of what is that column... CV name or input name? It would be nicer IMO to not have it at all, until you scroll down to that line... and maybe then alternate between the two. I think something similarly weird happens if you are using flight modes... i.e. there have been some interesting compromises made for the narrower screen B&W screen layout. Something to think about for the future though. 

Yeah, sometimes have to watch it when pressing... if there is switch bounce you'll get the get the up state instead... anyway, glad it's all working for you now :)

closed 2022-11-22 by pfeerick