EdgeTX/edgetx #2753: Frsky Neuron telemetry not restored correctly by Companion

AlanP10, 2022-11-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Setup the telemetry for a Frsky Neuron 80 on a Taranis x9D+.
Everything works fine, values for the following are recorded correctly EscV, EscA, BecV, BecA, EscR, EscC
Read the model from the transmitter to Companion then write the model back to the transmitter.
The following telemetry values no longer show the expected data: EscA, BecA and EscC.
EscA shows the same value as EscV.
BecA shows the same value as BecV
EscC shows the same value as EscR.
 

### Expected Behavior

The telemetry values should be the same after being restored from Companion

### Steps To Reproduce

1. Setup the telemetry for a Frsky Neuron 80 on a Taranis x9D+.
2. Read the model from TX into Companion
3. Write model from Companion to TX

### Version

2.7.1

### Transmitter

FrSky X9D+

### Anything else?

The YAML files are different.
The value of SubId is 1 for EscA, BecA and EscC after being restored by Companion it has a value of 0.
The telemetry interface on the transmitter doesn't allow SubId to be set so it's awkward to overcome this issue
[model02-correct.txt](https://github.com/EdgeTX/edgetx/files/10032504/model02-correct.txt)
[Model02-err.txt](https://github.com/EdgeTX/edgetx/files/10032520/Model02-err.txt)

AlanP10, 2022-11-17:
A workaround is to edit the YAML file after Companion has written the model to the TX.
The YAML file is in the MODELS directory on the TX SD card.
Notepad++ works well on MS Windows.
Change SubId to 1 for EscA, BecA and EscC

elecpower, 2022-11-17:
@AlanP10 can you please zip and upload the model file from the radio before it is imported into Companion as I need to see all the settings before your manual edit. Thanks

elecpower, 2022-11-17:
@AlanP10 ... or is the model02-correct.txt the original from the radio?

AlanP10, 2022-11-17:
Hi,

model02-correct.txt is the YAML file.
The bug report wouldn't allow me to upload a .yml file.
The file on the TX is model02.yml

Regards
Alan

elecpower, 2022-11-17:
Triage: issue affects all sensors with subid > 0

PR #2756 (2022-11-18): cpn(fix): yaml encode/decode telemetry sensor subId
Fixes #2753 though applies to all sensors
closed 2022-11-18 by pfeerick