EdgeTX/edgetx #2754: FrSky R9M-2019 won't bind on X10S / X12S radios

daveelsey, 2022-11-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I installed EdgeTX on both X10S-Express and X12S Access radios. Everything works fine except the 900 mHz module will no longer bind to any of my receivers. The 2.4G binds normally...  Switched back to OpenTX and the 900 mHz works just fine.
Nice work otherwise.

### Describe the solution you'd like

A firmware fix.

yak-54, 2022-11-30:
Hi i have X12S R9M 2019 with FW-R9M-ACCST-20190117 FLEX firmware on the module 
with R9Mini  on FW-R9MINI-ACCST_v20190201  FLEX firmware on it works fine

But if your using R9M 2019  with ACCESS i don't think its supported by Edgetx  on the External module bay yet

daveelsey, 2022-12-01:
Yes, I am using ACCESS - 2019 and not anxious to change back to ACCST.
Thanks for info.

yak-54, 2022-12-02:
so have you done external access mode to your X12S ?

daveelsey, 2022-12-02:
Yes, I did. Works perfectly with Opentx.

Capt Dave

raphaelcoeffic, 2022-12-10:
@daveelsey it was working for me as well last time I tested. Which FW version did you compile? Please note that if you did not compile a special version with ACCESS mod support, this is probably the reason it did not work. Alternatively, you could test this PR that allows for run-time configuration of the mod: #2562

yak-54, 2022-12-10:
> Yes, I did. Works perfectly with Opentx. Capt Dave
> […](#)

thats odd i have it working r9m access and r9m on expressLRS as well

daveelsey, 2022-12-10:
> Yes, I did. Works perfectly with Opentx. Capt Dave
> … <#m_5985140491773639392_>
> On Fri., Dec. 2, 2022, 3:22 a.m. Colin, *@*.*> wrote: so have you done

Thanks but how do you explain the X10S-Express having the same problem?
(OEM Access) I have not tried this and won't have time till this winter.
I looked at theX10S and the X12S in OpenTX and there is no "Serial Port"
section in Hardware. I guess that is unique to EdgeTX.
https://github.com/EdgeTX/edgetx/pull/2562
Raphael, does this help?   What firmware are you using?   X12s; Bootloader
2.3.14 otx, FW OTX x12s 2.3.14-otx (1e09791a), Modules / Rx version
ISRM-S-X10S  1.1.0/2.1.6FCC, Ext module OFF.

daveelsey, 2022-12-10:
Both my R9M-2019 modules are V1.3.0 FW, not sure why they show up as
V1.1.0/2.1.6FCC on the radios.

daveelsey, 2022-12-10:
Correction, the X12S's Ext module is ON, not off.  R9M  256.15.0/1.3.0 FCC

yak-54, 2022-12-10:
your right i thought i did 2.8.0 test it on Access 
it works with Accst only there is no where  in the External Rf menu to set it to Access 
or in the Hardware menu to there is no External RF sample mode for Access 

so me it looks like 2.8.0 issue they let it out

daveelsey, 2022-12-10:
Thx Colin. Most of this stuff is over my head, but I try...

Capt Dave

yak-54, 2022-12-10:
its up to developers to fix it i am in the same boat as you will not work :)

raphaelcoeffic, 2022-12-11:
> it works with Accst only there is no where in the External Rf menu to set it to Access or in the Hardware menu to there is no External RF sample mode for Access

I can only repeat the previous statement: you need the following to be able to use ACCESS on the R9M on radios that are not "native ACCESS":
- physical ACCESS mod
- and a custom built firmware (or use the firmware from #2562)

If you cannot select the proper module ("R9M ACCESS"), then it means that you don't have the proper firmware.

daveelsey, 2022-12-11:
Raphael, I had installed EdgeTX V2.8 on both radios. Where exactly can I
find the "Custom built firmware" you speak of ?  I have studied #2562 and
I'm not at all clear what you mean.

*"This PR removed the need for a custom firmware in EdgeTx. It allows you
to configure this ACCESS mod at run-time rather than requiring a custom
firmware, which would be required otherwise."*
*"Regular EdgeTx firmware does not support ACCESS mod. You must use a
custom firmware." * ?? They seem to contradict each other.

Do I re-install V2.8 and change the AUX1 settings in Hardware, as shown in
the pictures? Done?
or
Do I find a new "Custom" firmware V-? somewhere, and install it, without
doing the above action?

I apologize that my skills aren't adequate but I need it dumbed down a bit.
:)
Thankyou for all your help and dedication to this project.

rotorman, 2022-12-11:
As auto ACCESS mod PR is not yet merged, you have presently the following options:
a) you can build your custom firmware, e.g. with GitPod (see https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod ), or
b) download the firmware from not-yet merged PR #2562 Action artifacts (you need to be logged into Github to be able to download the artifact zip).

closed 2022-12-14 by pfeerick

daveelsey, 2022-12-14:
Thankyou everyone but "a" is way way over my head, and "b" I have
logged into Github PR 2562 and can't find anything that resembles "action
artifact"??
To be fair, I haven't tried reinstalling Edge and simply change the serial
port settings to see if that alone works.
Hopefully this mod will be simplified and incorporated into EdgeTX so us
regular folk don't have to go through all this.
Take care

rotorman, 2022-12-14:
@daveelsey PR #2562 was merged 5 hours ago, so you can find the code of it being integrated from tomorrows nightly.