EdgeTX/edgetx #2758: Companion 32 bit: Unable to download meta data Error:99 TLS initialization failed

Dweinbach, 2022-11-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

when I try to download components or check for updates on my old windows 10 32 bit computer, I get an error - 
Unable to download Meta Data.
Error:99
TLS initialization failed

tried uninstall/re-install - same result.

Edition	Windows 10 Home
Version	22H2
Installed on	‎6/‎23/‎2020
OS build	19045.2311
Experience	Windows Feature Experience Pack 120.2212.4190.0

### Expected Behavior

should be able to check for updates on my older 32 bit laptop

### Steps To Reproduce

1. install companion 2.8.0RC4 Flying Dutchman on 32 bit windows pc with windows 10 home
2. configure companion to download components/check for updates
3. try to check for updates or download components - error message appears

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.8.0.RC4

elecpower, 2022-11-18:
@Dweinbach usually (but not in every case) caused by an incompatible version of the ssl libs being found on you computer ie incompatible with Qt version bundled with installer.
The Companion installer should be copying compatible libs into the application directory. At runtime this is the first place Qt searches and from there a more convoluted search regime.
You stated you uninstalled and reinstalled. I would suggest you uninstall Companion. If the install directory still exists and dlls present this is likely to be the issue, if so delete the remnants and reinstall. If no remnants then either some other process is loading the incompatible libs prior to Companion and so the OS finds them in memory or some other override is in play like PATH variable.

elecpower, 2022-11-25:
@Dweinbach have you been able to solve?

MarSik, 2022-12-22:
I have the same issue with Linux AppImage in the final 2.8 on Fedora 36 (Same radio, but that is not relevant to the report)

The app prints out:

```
[W] unknown():0 - [qt.network.ssl] QSslSocket: cannot call unresolved function SSL_get_peer_certificate
[D] UpdateInterface::reportProgress():339 - "Unable to download Meta Data.\nError:6\nSSL handshake failed"
```

elecpower, 2022-12-22:
@MarSik thanks for the debug info. Appears to be a ssl support library issue in that Qt cannot find a compatible library and version. This error shouldn't be occurring for you as the compatible libraries should be bundled in the AppImage.

elecpower, 2022-12-22:
I just pulled apart the 2.8.0 Appimage and discovered the openssl libs are not bundled.
Ran the Win 32 bit installer and not included there either.
Ran the Win 64 bit installer and they are included.

MarSik, 2022-12-23:
@elecpower Thanks for checking. Installing `openssl1.1` in Fedora fixes this issue indeed. But as you said. AppImage should be self-contained.

elecpower, 2022-12-23:
@MarSik thanks for the update much appreciated

closed 2023-02-22 by pfeerick