EdgeTX/edgetx #2765: PPM Center Def/Max Buttons Don't Register

jmxp69, 2022-11-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EL18 Nightly (f84c26b6)
Pressing Min/Def/Max on the screen registers a screen touch evidenced by haptic feedback; however, the PPM center value does not change with every press. The button highlight also does not switch to the button pressed on every registered keypress.

### Expected Behavior

Min/Def/Max should change PPM center value on every registered touch screen button press.

### Steps To Reproduce

Edit any output
Highlight PPM Center
Press Min/Def/Max randomly.

Min/Def/Max do not change the value with every press even though the touch screen sees the request (haptic feedback occurs with every touch). If they appear to respond initially, I can usually force the issue with a +/- press, then min/def/max.

### Version

Nightly (Please give date/commit below)

### Transmitter

Flysky EL18

pfeerick, 2022-11-22:
Ok, this is a somewhat different issue to that intially described - it seems to have nothing to do with the DEF/MAX buttons when on this page, but actually more generally - i.e. if you bring up the number edit keypad on the mixes page it will also freeze every so often in min/max/default, and seems to be more prone to happen on the EL18 (and probably NV14) - I've not been able to reproduce it on the TX16S yet. It's an intermittent issue that seems to be resolved on a second press in most cases though, so may be a timing issue with the touch driver, which I think is supposed to get redone for 2.9, so will tentatively earmark it for that. 

A secondary aspect to this is that you will get haptic or sound feedback if configured for all regardless of where you touch, or if it is a valid input - it simply a beep/haptic tick to acknowledge there was some input - much like a key will click even if it does nothing.

gagarinlg, 2023-05-05:
I can not reproduce this with the current nightlies

gagarinlg, 2023-06-29:
closing this issue, can not reproduce

closed 2023-06-29 by gagarinlg