EdgeTX/edgetx #2781: 2.8.0 RC4 Flight Mode fade in/out jumps to zero

sam2b, 2022-11-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

_See attached file below in Steps to Reproduce._
Note: I am using 2 additional flight modes so their trim settings are relative FM0.

**Scenario:** On an airplane, when switching between FM1 and FM2, and switching from FM1 to FM0, and having these flight modes assigned to individual flap mixes, and enabled Fade In or Fade Out of 1 second.

**Behavior:** The flaps jump to zero before traveling to the intended position.
This can be seen both in the Companion Simulator and the physical servo movements.

### Expected Behavior

The flap movements should move from it's relative position to the new position as specified in the respective mix.
Whatever the case, the movement should not "jump" to zero before moving to its new position as specified in a mix.
I'm looking for smooth movement behavior between 3 or more mixes when used with respective flight modes, please.
The idea is so I can have individual trims for each of these settings so the plane remains in trim in any flight mode / mix.

### Steps To Reproduce

1. Please see the attached zip file containing the example model, and use SA switch to toggle between flight modes FM0, FM1, FM2 respectively.
2. In the outputs menu (channels 1 and 6) observe the flaps (flaperons) jumping to zero between flight modes and respective mixes.
[FlightModes.zip](https://github.com/EdgeTX/edgetx/files/10070300/FlightModes.zip)

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Version: 2.8.0 RC4
_See attached file above in Steps to Reproduce._
If necessary, I can make a video to show the physical behavior, but the Companion Simulator shows this behavior clearly too.

pfeerick, 2022-11-23:
Looks like you need to remove the switch assignment in the mixes...  at (1)

[image]

As those mixes are already enabled/disabled by the flight modes (2), by also having the replace mixes on switches, they are effectively being turned on and off at the moment you change that switch position... so the flight mode doesn't have anything to transition from since it would start from the center again.

closed 2022-11-23 by pfeerick

sam2b, 2022-11-23:
Thank you so much! It now works wonderfully.