EdgeTX/edgetx #2783: Zorro's Special functions page does not show switch name

andreo32, 2022-11-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On Special function page, switch name used for functions are not visible only switch direction.
[image]

### Expected Behavior

I want to see switch (A-H) name before switch direction which enable function.

### Steps To Reproduce

Open model menu and go to 9th page - Special functions.
Try to create one function which uses one of the switches for enabling this function. 
The switch name does not appears in first column. Only switch direction. 
Note: selecting proper switch by switch movement works as expected, only the switch name is not seeing.

### Version

Other (Please specify below)

### Transmitter

Radiomaster Zorro

### Anything else?

Version 2.8-RC4

pfeerick, 2022-11-23:
Can you check what SYS/RADIO -> HARDWARE shows for the switch names... if it's not showing `---`  or a custom name, but instead `   ` then for some reason your switch names are spaces, and hence a blank name is being shown.

[image]

If this is the case, it would be good to know how you got to 2.8.0-RC4, as this has been an intermittent issue (blank switch name) that we've not been able to pin down the cause of.

andreo32, 2022-11-23:
I upload firmware using Chrome applet [https://buddy.edgetx.org/#/flash?version=v2.8.0-rc4&filters=includePrereleases](https://buddy.edgetx.org/#/flash?version=v2.8.0-rc4&filters=includePrereleases) described here in [GitHub](https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0-rc4). It was smoothly upload. The reason for this update was date and time of TX which is reseted after every switch off/on of TX. I thought that might help (but unfortunately not)
[image]

Back to switches.
Until my firmware update, switches were seeing well. Now, under SYS/Radio->Hardware page they are seeing normally too and I have no customized names:
[image]
I checked other pages and the same issue appears also on Logical switches page
[image]
and subpages: edit Input record where command is managed by a switch (see the Source line in a picture below)
[image]
and edit subpage of Logical switch too
[image]

I do not know if this is connected but Companion properly exposes switches on all pages.

rotorman, 2022-11-23:
About date/time - check the internal CR1220 battery, possibly this is empty (it powers the real-time-clock).
About remaining issues - can you do a clean new microSD card (with Buddy) and try with a brand new model if the other issues are gone?

andreo32, 2022-11-23:
Date/time internal battery is OK - at least Calibration page shows that. I do not want to break the guarantee sticker to measure its voltage directly.
I will try to recreate new SD card this evening.

pfeerick, 2022-11-23:
> Until my firmware update, switches were seeing well. Now, under SYS/Radio->Hardware page they are seeing normally too and I have no customized names:

No, that's not normal - there should be three dashes to indicate the name is blank - somehow you've ended up with spaces in the name. Is there any chance of getting a copy of `RADIO/radio.yml` file before you make any further changes to the radio? 

[image]

By "know how you got to 2.8.0-RC4" I was meaning more from what version you updated from, as well as the how. Also, has Companion been used at all (to include or exclude that from the mix)?

andreo32, 2022-11-24:
Thanks @pfeerick for reply. Very informative! I started to use EdgeTX/OpenTX a few months ago, so I'm pretty new here.
I tried that action, which @rotorman proposed to recreate a new SD card from scratch. And indeed, it helps. So what I did and found out:
I took newly formatted SD card and put latest release on it by Chrome [(SD card content install)](https://buddy.edgetx.org) as described under installation guide. I manually copied just _radio.yml_ file to new SD card from old SD card. Switched on the TX and created one new model with one dummy request with one switch. The name of switch was exposed normally with switch direction as expected.
Next, I opened Companion (v2.8 as I had 2.8-RC4 firmware) and tried to copy models from previously generated Model and settings file. And here we are. The switch's names were damaged again.
So, it must be something on Companion side. I tried again, this time a little different approach: I recreated SD card again, put _radio.yml_ file and all _model's files_ (from Models directory) into new SD card. That was OK too until I update them using the same old Companion _Model and setting file_. So, a little closer - it looks like Companion's _Model and setting file_ is damaged.
Third approach was almost the same, but instead of updating TX from Companion's file, I first read current models and settings from TX and save them as **new** _Model and settings file_. This way was OK also after Companion's update of TX.
I was wondering why and when this happened. It looks like v2.8-RC4 is more restrictive or read that information from different source, because before upgrade I didn't have such issues. However, why happened I do not know - I didn't remember exact procedures I took beside upgrade. All I remember is, that I had sometimes problems ejecting USB connection of TX from PC. That time (again sometimes not always) I got damaged models directory (I restored them from backup).

Ha, another thing I found out during this analyze: the date and time suddenly do not reseting after every switch off/on of TX.

Conclusion:
My opinion is that, _radio.yml_ file got damaged once in the past with one of the procedures of configuring it using Companion. How, I have no idea.

Thanks all for help and all information I got.

PS: I attached also Companion's Models and Settings file if someone has time to analyze them.
[2783_issue.zip](https://github.com/EdgeTX/edgetx/files/10081787/2783_issue.zip)

pfeerick, 2022-11-24:
If that is what was used to edit the first SD card, then yes, you're basically re-writing the same ... not necessarily damaged or corrupted... let's say *misconfigured* for want of a better word ... configuration to the radio, hence re-introducing the problem again. But, looking at both the `radio.yml` and the config in Companion, it is what I suspected... three spaces, hence the firmware is showing the name configured.

[image]

Basically, open that config again in Companion (or preferably read it from the radio - really should be reading from the radio, making changes, and writing back, so changes you make on the radio aren't lost), and remove the spaces from the switch names, and then write it back to the radio. That should be the end of the issue. 

Do you remember what version you had on before you upgraded? 

https://user-images.githubusercontent.com/5500713/203735388-4561a8d6-de44-4bc5-8c2e-239feb1f72f4.mp4

andreo32, 2022-11-24:
I bought Radiomaster's Zorro which had version 2.7.0 installed. That time version 2.7.1. was officially released, thereby I upgraded it to 2.7.1.

Aha, _radio.yml_ is a text file. Cool. Well, I never looked into it.
I never check - actually it is difficult to see if you do not take procedures like on the uploaded video, for spaces in the names. Although, regardless of those spaces, I saw the switch's names on TX with version 2.7.1. Obviously, this is one of the changes on 2.8 version.

Is it possible to export Companion's _Model and settings file_ (file of type .etx) into text file or at least each model and/or radio settings?

Warmly regards,
Andrej

andreo32, 2022-11-24:
I found that Models can be exported. What about radio settings?

pfeerick, 2022-11-24:
Ah, thanks for that. It's not an intended change ... so this suggests something must sometimes be going wrong when updating from 2.7 to 2.8.

Try 
[image]

Another think to take note of is an `etx` file is actually just a zip file with a different extension, with the same structure as the radio - so has MODELS and RADIO folders, etc.

andreo32, 2022-11-24:
Thanks @pfeerick far all information! I will check than, when this happened in my backup. Maybe I will find.

andreo32, 2022-11-24:
Hm. It was on 6th of November. So far away from upgrade. I do not remember what I was doing that time...
And the other strange thing is that now, after SD card recreation from scratch, date and time is correct. 
So we solved two issues at once 🥇.

closed 2022-11-24 by andreo32

Roman-Golod, 2022-11-29:
I can confirm this issue. I've had this problem in the 2.8 rc3 upgraded from the 2.7.1. I can't remember if it was in the 2.7.1. I discovered spaces in the switchers' names. Also I have issue with the date/time resetting ([#2187](https://github.com/EdgeTX/edgetx/issues/2187)). For now I can't say if it gone, I've flashed 2.8.0 just yesterday. On the left side of screenshot - 2.8.0 from scratches, on the right - models.yml from my Zorro.
[image]

andreo32, 2022-11-29:
Regarding date/time reseting. It must be something with Companion writing configuration to SD card, so I didn't  solve that issue. All as I remember, once I recreated SD card from scratch the date/time remains after switch off/on the TX. After that I change some configuration on my model using Companion. I didn't check the date/time after that. But at the field, a day later, the date/time was reseted.
I will try to recreate SD card from scratch again and will check how much time will date/time remain proper one.

TerraMotoEU, 2022-12-02:
> Regarding date/time reseting. It must be something with Companion writing configuration to SD card, so I didn't solve that issue. All as I remember, once I recreated SD card from scratch the [...]

I don't think the date/time reset is related to companion, or at least not exclusively. I've just received a brand new Zorro 4-in-1 and flashed it to 2.8 without ever having the companion used or even installed. Yet at first it resetted the date to something 1970 and later that resetting changed to year 2000. 
Also S1 lost its name and Rudder stick as well. I just gave them their names back and learned to ignore date and time altogether. Not showing it on my screens anyway, as all I need is my timers.

andreo32, 2022-12-02:
@TerraMotoEU , date and time is not so important until you want to log data.
Log file names are composed from date and time.
So imagine that after five days of flying you have a bunch of files with almost the same file name...
Last days I'm not able any more to recreate such SD card (like before) that date/time is preserved.

TerraMotoEU, 2022-12-04:
@andreo32 You're right. I just realized that it's messing with date/time stamp for model backups, too. Wasn't thinking of writing log files, but if I did I'd be heavily annoyed by messy file names.

pfeerick, 2022-12-04:
It's rather strange - Using Companion should have no effect, as the time is not saved to file, but to the onboard RTC (real time clock). If the time is lost for whatever reason, it would in the first instance reset to 1970, and then when the code restarts the RTC (since it's not set/running at all), it will jump to 2000... so you should see on one power up it be 1970, and on the next 2000 (if you don't set it in the meantime, of course). I would normally only expect to see the time being lost if the RTC battery goes flat... but there are too many reports of it being lost, so I suspect something else is going on here... as it should be a year or so before I would expect to hear reports like that (and that replacing the 3v backup battery fixed it).

gagarinlg, 2022-12-04:
Sounds a bit like a power glitch when connecting or disconnection USB.

andreo32, 2022-12-04:
@gagarinlg, 
It is possible that glitch reset the date/time, yes.
However, date/time reset happened also on regular switch on/off - without connecting TX to PC.
I'm just wondering, is it possible that RTC is resetted on some special condition during switch on or off the TX?

gagarinlg, 2022-12-04:
Nobody should reset the RTC.
There might still some glitches happening by just switching on or off. I do not have a Zorro to check, so someone else has to do this. But this issue is closed, can you please move over to the Zorro RTC ticket for the discussion?