EdgeTX/edgetx #2786: IMU not working properly on X12s still!

Kevltan, 2022-11-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The IMU problem is back again!
I am re-opening this issue because I believe there is still something wrong here. What I have noticed now is IMU won't work all of the time on power up, but if I hold the T1, T4 trims in and and power on as though I was going to do a firmware update in bootloader and then exit instead, it works every time! Anybody any ideas? I really am giving up here because there seems to be no logic in it. I think I've fixed and then I haven't! Still thinking there must be something in the startup/powerup routine that is different to the boot loader and exit routine - works everytime from the bootloader🤔.

### Expected Behavior

IMU not functioning every time radio is switched

### Steps To Reproduce

1. Sometimes when powering up no IMU active.
2. No response until a power recycle - maybe....sometimes!
3. Power up with bootloader and exit IMU functions everytime.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

### Anything else?

2.8.0 selfbuild RC4

pfeerick, 2022-11-24:
Raphael has commented in the dev chat that there is a clash between touch screen controller task and imu task, so this may be related, and as such is on the radar to be fixed, but that will most likely be a 2.8.1 fix.

Kevltan, 2022-11-24:
Nice one, glad to here this Peter. Can live with at the moment but look
forward to the fix in 2.8.1 🙂👍

Kevltan, 2022-11-28:
I am having no problems with IMU since changing to 2.8.0 "Flying Dutchman" now working everytime. Has something been changed? I have noticed that there is no call for IMU_LSM6DS33 in System/information/version/OPTS:, which I called for in
cmake -Wno-dev -DPCB=X12S -DDEFAULT_MODE=1 -DIMU_LSM6DS33=YES -DCMAKE_BUILD_TYPE=Release ../. this was shown listed in the my OPTS: Had I created the bug myself when self building?
The absense IMU call in the list seems to have no effect of IMU's operation which now seems to work as it should! 🤞🤞

pfeerick, 2022-11-28:
That is quite possible... Unless you're using the LSM6DS33, `IMU` is the flag for IMU support on the X12S (and is on by default)... ;) 

[image]

Kevltan, 2022-11-28:
The only time I can the IMU to freeze at the moment is to edit "screen settings" i.e remove and add TltX or TltY - a power re-cycle returns it to working condition again though.

Kevltan, 2022-11-30:
There are still problems with the IMU freezing when going through various model/system pages, but it can be resolved with a power re-cycle now.

1Sven, 2022-12-01:
A big problem, because IMU and display interfere. Raphael already has a great idea and I hope he solved this bug in 2.8.1 as well.

Kevltan, 2022-12-07:
I temporarily disconnected P8 and I2c wiring from the touch panel to eliminate all hardware additions. Tried IMU functionality and still failing with the same shortcomings as decribed earlier. So believe this is definately a firmware problem/conflict somewhere and not hardware related.

Kevltan, 2022-12-14:
I temporarily disconnected P8 and I2c wiring from the touch panel to eliminate all hardware additions. Tried IMU functionality and still failing with the same shortcomings as decribed earlier, this isolates the touch panel itself from the equation. So believe this is definately a firmware and/or SD card software conflict somewhere and NOT hardware related.
I have reloaded the themes software on the SD card because I was losing the IMU when editing the colours. This seems to have "improved" the issue and so far have had no failures going in and out of menus I mentioned previous, maybe the conflict resides in the graphics driver or themes software somewhere!! Will keep on testing to see if problem in any way returns🤞🤞🤞

Kevltan, 2022-12-14:
Some interesting information I found which might help the developers pin point the problem. I have 2 SD cards one 8GB and one 16GB, both formated and named the same with EXACTLY the same software copied on them. The IMU does not work with the 8GB but does with the 16GB when TX booted, but they both load with everything else working normal. This is 100% repeatable.....Any ideas??

Kevltan, 2022-12-25:
Still experiencing same problems, IMU sometimes works and sometimes doesnt when powering up or during editing it freezes. Always works when initiating boot loader and immediatley exiting using the exit prompt but after this can still freeze whilst editing pages. This is seems to be NOT related to a hardware probem as is still repeatable with the touch panel disconnected.

pfeerick, 2022-12-29:
Only someone with access to the hardware is likely to be able to fix it. @raphaelcoeffic is one such person, and as was commented on earlier, has some idea as to what could be the issue ;) As far as whether there is a historical problem, that is anyone's guess atm, as since you point it has had limited use. However, I'm inclined to think it is still related to the TP support, regardless of if one is connected or not, as that is fairly new and an unusual combination.

Kevltan, 2022-12-29:
OK that's great, so thinking it is a clash/conflict in the firmware. But
finding it strange that it works everytime from the bootloader and exit🤔

Kevltan, 2023-01-01:
Do you know if the X12S IMU issue has been resolved as it seems to be working fine since the last nightly (01-01-23). So far, not a fail from power up, and I've done it about 40+ times. The last nightly did not work at all unless from bootloader and exit! Has something been changed?

Kevltan, 2023-01-05:
I have managed to get it to work consistently now on power up. Seems to have been a conflict between the my own created splash.png and the power up cycle. It also works fine with the EdgeTX default splash.png. I found by taking my splash screen into Paint.net and re-saving it as a .png again the conflict has disappeared! the other thing I have noticed is it wont work with a standard SanDisk SD card less than 16GB, it works fine with my Kingston 16GB SDHC XCI class 10, but does not work with a SanDisk Ultra 32GB SDHC I class10! All cards definately do not seem equal here for some reason and I have brand new Lexar 32GB SDHC I cards that dont even work at all and report back Bad Data no matter how I format them and work fine in other applications......🤔🤔

Kevltan, 2023-01-07:
All problems with IMU not working on start up caused by having a splash.png file too large in the System/Images file on the SD card. The size was 220kb this is too large and seems to upset the IMU on start up. I used an online .png file compressor to reduce it to 89kb without changing pixel count of 480 X 272 https://tiny-img.com/png-compressor/. I have tried this on cards from 256Mb - 32Gb standard and SDHC format. All formatted to Fat32 default allocation. IMU has never failed. So if your running your own splash screen at start up i think in cant be over 100kb in size. The ones created online in Air-RC.com or Skyracoon.com are too big and need reducing.
Hope this helps anybody experiencing this problem.
I only wanted the IMU for screen wakeup!!!!

Kevltan, 2023-02-02:
THIS NOW WORKS FINE AFTER THE ABOVE ISSUE ADDRESSED

raphaelcoeffic, 2023-02-10:
@Kevltan the issue with picture size / time to load is now obvious. The sensor initialisation is done in the UI task, while it is queried in the mixer task. This can never work and basically amounts to a race condition as which one of the two happens first:
- error counter in mixer task is incremented up to 100, at which point the sensor will not be asked again,
- and when the sensor is actually initialised (hopefully before the error counter reaches 100)...

Kevltan, 2023-02-10:
Ok so would the answer be to limit the image file size of the Splash.png
and if it is to big, it is thrown out before startup?

raphaelcoeffic, 2023-02-10:
> Ok so would the answer be to limit the image file size of the Splash.png

No, you don't take aspirin when you have a bullet in your head: you go to surgery. Here it means, we fix it properly (see 828b5c0c25e2df7e75177d9e63b194ea857630dc).

PR #3171 (2023-02-15): fix: various issues related to IMU sensor
Summary of changes:
- fixed a lock-up issue with touch sensor and IMU on the same I2C bus
  - Please note that the HAL already handles busy I2C bus, so that we should not run into the case where the bus is used and another query is started. However, if that happens, HAL_BUSY will be returned, which leads to -1 being returned in our functions, thus increasing the error count until the sensor stops working.
- fixed a race condition between init and usage of the IMU sensor
  - Init was placed in the UI task, while the sensor was queried in the mixer task. [... cut]
closed 2023-02-15 by pfeerick