EdgeTX/edgetx #2788: input and mixer screens does not hightlight which one is active

wookie457, 2022-11-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

just flashed 2.8 when in the input or mixer screen and for example have three rates set up all three are white when switch is moved, it will not highlight the active rate or mixer

https://user-images.githubusercontent.com/57927112/204141016-38a398df-8057-49f2-8af6-de88f6bd51d0.mp4

### Expected Behavior

active input or mixer highlights when selected

### Steps To Reproduce

 1. set up multiple rates in input screen or mixer screen
 2. set switch to control rate
 3. move switch while monitoring for changes

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.8 version

gagarinlg, 2022-11-27:
This must be a problem of the theme you are using.
Check the "SECONDARY2" color, it has to be different from "PRIMARY2" for the highliht to work.

apart from that, maybe we should use the "ACTIVE" color for the broder of the active mixers. @raphaelcoeffic, @JimB40 ?

wookie457, 2022-11-27:
Thank you very much. Yes the 2 colors on the theme were the same. I adjusted and now get the proper highlighting.

gagarinlg, 2022-11-27:
> Thank you very much. Yes the 2 colors on the theme were the same. I adjusted and now get the proper highlighting.

But you get the crown for being the first to create an issue for the released 2.8!

closed 2022-11-27 by gagarinlg

wookie457, 2022-11-27:
LOL, well thank you! But really 2.8 looks really awesome! You guys have done a great job!