EdgeTX/edgetx #2789: Jumper T-Pro not work on v2.8.0

hackerboygn, 2022-11-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I upgrade from 2.7.1 to 2.8.0, Jumper T-Pro can't turn on. Then I downgrade to 2.7.1, it's work.
I flash bootloader to 2.8.0, It's can into new bootloader, but will I plug in the usb cable, bootloader carshed.
I format and overwrite TF files of 2.8.0 contents.

### Expected Behavior

Work

### Steps To Reproduce

Jumper T-Pro upgrade to v2.8.0

### Version

Other (Please specify below)

### Transmitter

Jumper T-Pro

### Anything else?

Jumper T-Pro EdgeTX v2.8.0 Firmware.

pfeerick, 2022-11-27:
> I upgrade from 2.7.1 to 2.8.0, Jumper T-Pro can't turn on. Then I downgrade to 2.7.1, it's work.

How did you upgrade?

>I flash bootloader to 2.8.0, It's can into new bootloader, but will I plug in the usb cable, bootloader carshed.

Ok, so you have the 2.8 bootloader on, and get get into the bootloader. When you plug in the USB cable, and it crashed - how? Did it go back to lighting up the FS leds in indicate charging/charge level? If so, try putting it into bootloader again - it does that once in a blue moon - probably something to do with it being plugged in. Try plugging in the transmitter end first, and then the PC end second. Also, if you are using a USB hub, bypass that, and plug directly into the PC - the Jumper USB-C also charges the battery, and can draw too much power for a hub and make weird things happen.

Also, if you flashed the bootloader, you must have the firmware on the SD card already - you should be able to flash the firmware via the bootloader without plugging in the USB cable. 

DFU upgrade is final option - i.e. with the transmitter powered off, hold down the recessed button down at the top near the USB down and plug the cable in. Then use https://buddy.edgetx.org/#/flash?version=v2.8.0&target=tpro to flash it.

deadbytefpv, 2022-11-29:
[image]
Just updated mine. No issues here..

bigbear2001, 2022-12-01:
> [image] Just updated mine. No issues here..

[image]
Confirm

nptr, 2023-05-05:
I had the same as @josephduchesne encountered, just with flashing to `2.8.3` also from `2.7.0-fcf8fbd` (stock fw). 
_(Thanks for basically posting the solution for me!)_

1. Installed bootloader via the in-built menu
2. Booted into it (trim buttons inward + power)
3. Wrote 2.8.3 firmware
4. Restarted => No boot

Reinstalling the stock firmware from the 2.8.3 bootloader and then using the DFU method to flash `2.8.3` worked. A bit more complicated than the in-radio method but whatever. Not sure if its a 2.7 or 2.8.X problem. Or possibly an incompatibility between those.

pfeerick, 2024-03-04:
I believe there was some minor glitches around sizes of firmware around 2.8.x time, so given this could 'easily' be completely resolved with a DFU flash, which completely updates both the bootloader AND firmware, I think this can be closed now. Please open a new issue if this is still an issue with 2.9.x or nightly (2.10) firmware.

closed 2024-03-04 by pfeerick

zhuwei, 2025-02-14:
Although the firmware can be successfully flashed, there is no feedback from the two remote controls

pfeerick, 2025-02-15:
Is this in relation to the T-Pro? And after having flashing 2.10.x (as the current "stable" version) using DFU? Are you sure the right target is being flashed? As there is the T-Pro and the T-Pro V2.

zhuwei, 2025-02-18:
@pfeerick 
I just have the T-Pro V1 in my hand
Now the DFU is 2.8.0, but the firmware has been flashed to 2.6.0 by me. I don't know why the two joysticks still don't respond, while the other buttons are normal.
Before the upgrade, the joystick remained normal. It's too painful.

pfeerick, 2025-02-18:
> Now the DFU is 2.8.0, but the firmware has been flashed to 2.6.0 by me. 

Huh? Do you mean the bootloader is 2.8, and you have flashed 2.6 for some reason?

You probably need to go into SYS -> Hardware -> Calibration and redo your stick calibraton.

zhuwei, 2025-02-19:
> > Now the DFU is 2.8.0, but the firmware has been flashed to 2.6.0 by me.
> 
> Huh? Do you mean the bootloader is 2.8, and you have flashed 2.6 for some reason?
> 
> You probably need to go into SYS -> Hardware -> Calibration and redo your stick calibration.

Yes, the bootloader version is 2.8.5. Do we need to downgrade the bootloader back to 2.6? But it seems that there is no place to download the DFU firmware file for version 2.6.
Calibration is ineffective because the joystick has not responded

pfeerick, 2025-02-19:
This is not how DFU works. DFU functionality is stored in the MCUs read only memory. 

If you want to flash the EdgeTX bootloader back to 2.6.0 (to match the firmware version), go into SYS => SD, browse to FIRMWARE, and select the firmware file you flashed via the bootloader. You will have an "Flash bootloader" option in the popup menu.

zhuwei, 2025-02-19:
> This is not how DFU works. DFU functionality is stored in the MCUs read only memory.
>
> If you want to flash the EdgeTX bootloader back to 2.6.0 (to match the firmware version), go into SYS => SD, [...]

I'll give it a try, thank you.
It was normal in version 2.6, but it became abnormal when I upgraded it to version 2.8.5. I really can't find the reason