EdgeTX/edgetx #2794: Rework battery monitoring function

richardclli, 2022-11-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

These radios supports battery charging via USB/Wireless:
  NV14/EL18/PL18

However the battery charging handling is not nice enough, it should have all the features that a normal cell phone has.

### Describe the solution you'd like

1. When the radio is off and charging is activated (wired/wireless), it should show a "Power off charging screen"
2. Power off charging screen should show charging progress in % and will off LCD with a timeout that can be activated by screen touch/ stick movement, etc. events.
3. When charging is deactivated in power off charging screen, (unplug USB, remove wireless charger), the radio should switch off
4. When user press power button in Power off charging screen, the radio will switch on.
5. When the radio is on and charging is activated, battery icon will show charging
6. There is a function/screen that can check the current charging progress when the radio is on
7. When charging is deactivated with radio is on, the battery icon and charging progress should show correct information that the charging is stopped.
8. When user switch off the radio when charging is activated, the radio will show the power off charging screen.

richardclli, 2024-06-01:
@philmoz Need your help on the UI part

fxk666, 2024-06-25:
There is another bug, open source pl18 I plug in the USB to power the remote control when it shuts down, there is a high probability of entering emergency mode

richardclli, 2024-06-26:
> There is another bug, open source pl18 I plug in the USB to power the remote control when it shuts down, there is a high probability of entering emergency mode

I never encountered this with 2.10.x firmware, try to see if it related to USB power supply.  Some computers do not have enough current to power up the charging circuit properly and cause a blackout.

gagarinlg, 2025-01-24:
@richardclli is this still an open issue with the changes you prepared to join NV14/EL18/PL18 code?

richardclli, 2025-01-31:
Yes, this is noy done yet.