EdgeTX/edgetx #2795: RSSI Model Locator hangs EdgeTx 2.8.0

DavidLorentzen, 2022-11-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Open the "RSSI Model Locator", shows 0 db whether a receiver is connected or not, program appears to be hung. There is no way to exit the screen without shutting down the transmitter with the power button.

Using 2.8.0 release.

### Expected Behavior

Should be able to exit the app with RTN button or SYS button.

### Steps To Reproduce

Pretty straightforward from the Current Behavior info.

1. Open the Model Locator (by RSSI) in the Radio Settings | Tools menu
2. EdgeTx 2.8.0 hangs, cannot exit except by powering down the transmitter

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

eshifri, 2022-11-29:
Are you using the latest version of the  "RSSI Model Locator" ?

DavidLorentzen, 2022-11-29:
Possibly not. Is it a lua script? I was thinking it was part of the 2.8.0 firmware. My ignorance is showing.

eshifri, 2022-11-29:
Yes, it is a LUA script.

pfeerick, 2022-11-30:
I don't think it has an exit function of sorts, so to exit it you long press the RTN key (and generally this is how you exit any standalone Lua script that doesn't explicitly do any exit handling). With no RX connected, are you still getting the roughly once-per-second beeps? If so, it's not hung, just has nothing to show. 

When you do have a RX connected, have you done telemetry discovery yet (on the models -> telemetry page) so that the RSSI sensor is present?

DavidLorentzen, 2022-11-30:
Sorry, this was all on me. I thought I had copied the updated lua scripts to the Tx, but all I had there was files from 8/21. It's all better now. And I didn't know the long-press trick to exit. Works.

closed 2022-11-30 by DavidLorentzen

pfeerick, 2022-12-01:
Glad to hear it's all working for you now :)