EdgeTX/edgetx #2796: Input Channel Change Delay

Durahl, 2022-11-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Greetings!

I'm currently optimizing my setup for Running ( AND Programming the `DMD Controller` ) Tamiya RC Tanks which requires a peculiar setup for each case due to their intended use with Trimmers. Before I'd do all of it in the `Mixes Tab` creating a huge clusterfuck but have now attempted to solve this by moving parts of it to the `Input Tab` to reduce clutter in the `Mixes Tab`.

Technically, what I've planned to do **_IS_** working just not as I expected and with it performing different on the `Transmitter` vs. the `Companion App` I assume something being amiss on the former...

[Here's a Video of both cases](https://youtu.be/f2jCfEML_lM) ( Transmitter vs. Companion ).
As can be seen when switching from the "**_Default_**" `SB-` to the "**_DMD Programming Mode_**" `!SB-` it takes AGES for the change to happen on the Transmitter vs. it immediately happening in the Companion App.

Unfortunately my Theme is kinda on the bright side when it comes to seeing the change happening on the Transmitter and the W11 GameBar Recorder wouldn't record the Simulations Window so... Yea... Crappy Camera recording for both 😉

[image]

### Expected Behavior

For the Input Change to happen immediately on the Transmitter like how it is in the Companion App Simulation.

### Steps To Reproduce

No clue... I tried removing the Voice Recording just in case but that didn't do a thing either...

### Version

Other (Please specify below)

### Transmitter

FrSky X12

### Anything else?

[ZIP File containing the Leopard 2A7 YAML Config.zip](https://github.com/EdgeTX/edgetx/files/10115815/Leopard.2A7.zip)
The Horus X12S was modded with a TouchScreen ( in case it may be of relevance 🤔 ).

eshifri, 2022-11-29:
Any chance you have some additional functionality attached to SB switch? E.g. logging?

Durahl, 2022-11-29:
Nope, nothing that I'd know of.

For testing purposes, I just created a new Model with just that setting alone where `SB-` and `!SB-` would change the settings of two Input Channels at the same time and it again lags behind when making the change from `SB-` to `!SB-` **but not the other way around**.

Durahl, 2022-11-29:
Yea... I'm kinda stupid...

At some point I must have assigned `!SB-` to taking a Screenshot as a `Global Variable` which obviously wasn't visible from the Model Tabs... Unassigning it fixed it.

/sigh...

closed 2022-11-29 by Durahl

pfeerick, 2022-11-30:
> At some point I must have assigned `!SB-` to taking a Screenshot 

The number of times I've nearly started digging in the code, or nearly opened an issue because of the same thing (I have Screenshot set as a global function on a couple of radios, and on different switches) would amaze... 😆 It's so easy to forget after coming back to a radio after a few weeks...