EdgeTX/edgetx #2797: Check FM Trims - inconsistenct behaviour

ulfhedlund, 2022-11-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The Check FM Trims button is not consistent between BW and color radios. For BW radios the button is translateable and the current flight mode is included in the button name, whereas for color LCD radios the button text "Check FM Trims" is hard coded and hence neither translateable nor includes the current flight mode. 

### Describe the solution you'd like

IMO the behavior for BW radios is preferable, i.e. the Check FM Trims button should be translateable and include the current flight mode.

### Describe alternatives you've considered

1. all firmware work the same way as for BW radios
2. color LCD radios should at least have a "Check FM Trims" button text that can be translated

pfeerick, 2022-11-29:
1) the behaviour is the same for all radios - it just disables trims for 2 seconds - irrespective of the active FM or if colorlcd/BW radio.

2) agreed... it needs to be made translatable also - which is just a matter of replacing the "Check FM Trims" text at 
https://github.com/EdgeTX/edgetx/blob/a267520236b2e04f502cac582787061dafd2c05a/radio/src/gui/colorlcd/model_flightmodes.cpp#L327
with `STR_CHECKTRIMS`

Whether it needs to specify the active flight mode on colorlcd is up to personal taste IMO - it's necessary on B&W as it's at the bottom of the list, you can't tell the active flight mode... but you can on colorlcd so not strictly necessary.

ulfhedlund, 2022-11-30:
@pfeerick: I can add the translation. Any suggestions where to add the string in the translation files?

pfeerick, 2022-11-30:
`STR_CHECKTRIMS` already exists - that's what's used on the B&W radios ;)

ulfhedlund, 2022-11-30:
Well, then I'll definitely be able to fix this :)

ulfhedlund, 2022-11-30:
This is what it looks like on TX16S. The text is on two rows in an almost square button which isn't fully visible without scrolling. It doesn't great, I think, but that's just my opinion :)

Should it be left like this or a new translation string added?

[image]

gagarinlg, 2022-11-30:
we could make the button wider, or even give it a dynamic width.

ulfhedlund, 2022-11-30:
Does it have anything to do with the special characters in the TR_CHECKTRIMS? The hard coded text is longer but is displayed in one row.

gagarinlg, 2022-11-30:
yes, \012 is in octal notation, so it is a line feed character.

ulfhedlund, 2022-11-30:
Well, if the line feed is removed it doen't work for BW radios, since  the current flight mode is inserted in the middle for those in some clever way. I don't how to fix this with for color radios my limited programming skills, in any other way than to add another string for color radios. 

I can do that if you want me to, but perhaps there is a better way to do it and use the same string for all radios that you prefer and can implement.

ulfhedlund, 2022-11-30:
OK, should I then add a new string for color radios?

ulfhedlund, 2022-11-30:
I see. Then I propose to solve it by simply adding an if-else section in the translation files. OK?

#if defined(COLORLCD)
  #define TR_CHECKTRIMS "Kolla FL-trimmar"
#else
  #define TR_CHECKTRIMS CENTER "\006Kolla\012trimmar"
#endif

PR #2800 (2022-12-05): Check FM Trims translateable for color radios
Fixes #2797

Summary of changes:
Added translation for Check FM Trims button for color radios and separated the button text string for BW and color radios in the firmware translation files.
closed 2022-12-05 by pfeerick