EdgeTX/edgetx #2798: Cannot set a high baudrate (≥921k) to my External RF in ELRS v3.1.0

CCNUI, 2022-11-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm always frustrated when I set a high baudrate (≥921k) to my External RF.
Extranal RF module: ExpressLRS BETAFPV 2.4G TX Micro 1000mW
ELRS Version: v3.1.0 or 3.x.x-maintenance
EdgeTX Version: v2.8.0 (fed140e)
[image]

### Describe the solution you'd like

🤔

lesharodin, 2022-12-01:
Try this
https://github.com/ExpressLRS/ExpressLRS/issues/1952#issuecomment-1332843863

pfeerick, 2023-02-19:
This should be resolved as of 2.8.1. Please comment further this not the case and the issue can be reopened

closed 2023-02-19 by pfeerick