EdgeTX/edgetx #2799: 2 issues with 'Build-Instructions-under-Ubuntu-20.04'

nostromo123, 2022-11-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

With  clean Xubuntu 20.04 in a VM, I did a test run for building.

One note (minor issue):
after finishing firmware

[100%] Built target firmware
```
~/edgetx/edgetx_main/build-output$ mv firmware.bin edgetx_v280_X10_Express_lua-noheli.bin
mv: Aufruf von stat für 'firmware.bin' nicht möglich: Datei oder Verzeichnis nicht gefunden
~/edgetx/edgetx_main/build-output$ cd arm-none-eabi
~/edgetx/edgetx_main/build-output/arm-none-eabi$ mv firmware.bin edgetx_v280_X10_Express_lua-noheli.bin
```
Could someone please add the path here in the description.

There seems to be a problem with building companion, after the firmware has been generated. Neither working in 
both folders ... 'no rule'

```
~/edgetx/edgetx_main/build-output/arm-none-eabi$ cd ..
~/edgetx/edgetx_main/build-output$ make -j`nproc` companion28
make: *** Keine Regel, um „companion28“ zu erstellen.  Schluss.
~/edgetx/edgetx_main/build-output$ cd arm-none-eabi
~/edgetx/edgetx_main/build-output/arm-none-eabi$ make -j`nproc` companion28
make: *** Keine Regel, um „companion28“ zu erstellen.  Schluss.
~/edgetx/edgetx_main/build-output/arm-none-eabi$ 
```

Could someone please have a look.

br - KH

### Expected Behavior

Build Instruction working ;-)

### Steps To Reproduce

See errors posted ....

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

Building was done today with

~/edgetx$ git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx_main

and

Xubuntu has been updated today

gagarinlg, 2022-11-30:
In the second step you tried to compile from the arm-none-eabi directory, instead of the build directory and the companion can not be cross compiled to run on the radio 😉.

rotorman, 2022-11-30:
I updated https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04 to better match the current step to build firmware and native binaries.
@nostromo123 please check if OK now and if shout if I missed something. If all is OK, please close this issue.

nostromo123, 2022-11-30:
Thx!

OK now - KH