EdgeTX/edgetx #2804: Touch control can hangs the radio

richardclli, 2022-12-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Can be reproduced by following the procedures in the radio.  Tested on EL18 and TX16S and both can be reproduced.

https://user-images.githubusercontent.com/18414861/204994101-36aab16b-c105-4e96-a4cf-eee212abe1c8.mp4

### Expected Behavior

Touch will not crash.

### Steps To Reproduce

Tap and slide up down with a long list.

### Version

2.8.0

### Transmitter

Flysky EL18, Radiomaster TX16S / TX16SMK2

gagarinlg, 2023-05-05:
This still happens with the current nightlies. Interestingly the screen flickers at the same time. Maybe that is an LVGL feature, when the touch screen is touched for a long time, in case it is unintentional.

richardclli, 2023-05-06:
> This still happens with the current nightlies. Interestingly the screen flickers at the same time. Maybe that is an LVGL feature, when the touch screen is touched for a long time, in case it is unintentional.

I think there are conflicts between screen dimming logic and LVGL logic, should see if the controls can be centralized in the LVGL backlight driver to resolve the problem.

pfeerick, 2023-05-06:
>  Interestingly the screen flickers at the same time. Maybe that is an LVGL feature, when the touch screen is touched for a long time, in case it is unintentional.

IIRC, I saw that in another issue to do with scrolling in the MPM protocol lists - whereby a _continuous_ drag didn't reset inactivity/backlight timers, only touch-'n-release.