EdgeTX/edgetx #2805: Frsky X9E Hall radio hangs after writing models and settings

Bratski, 2022-12-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

I flashed the Frsky X9E Hall-version Radio with the EdgeTX2.8 Firmware. After writing my models and settings in Companion 2.8 to the radio and removing the USB-Cable (yes I unmounted the file system first), the radio hangs. 

I have to disconnect the battery, and delete all of the models from the SD Card to be able to boot the system again.

### Expected Behavior

Maybe an error message, but not blocking the system completely.

### Steps To Reproduce

flash radio firmware to edgetx 2.8
run companion 2.8
connect radio, read models and settings
copy paste models from edge 2.7 savings
write models and settings to the radio
close and unmount
pull the USB cable
radio system hangs

### Version

2.8.0

### Transmitter

FrSky X9E / X9E Hall

Gerold68, 2022-12-02:
Same experience with my 9XE without Hall sensors.
I made this e. with 2.6, 2.71 and 2.8.
The radio works only if I create the models with the radio.
If I copy models via companion, the radio crashes when it boots the system.
Only deletion of Radio and Model yamls brings the radio back to life.

Gerold68, 2023-01-07:
Bratski,

please check if your issue is the same like mentioned in #2855 and #2950
Put your SD-Card into your PC and edit all modelxx.yml.
Look for the entry switchWarningState: and set it to "".
The Switchwarning function is then off, but the radio works again.
You help is appreciated.
Regard Gerold

Bratski, 2023-01-22:
Hello Gerold,
thanks a lot, that did the trick, sofar my X9E runs stable...
Best regards,
Bart

Gerold68, 2023-03-11:
This one is also fixed by #3232

closed 2023-03-11 by gagarinlg