EdgeTX/edgetx #2806: Out of order mixes do not appear in mix screen but send output

jmxp69, 2022-12-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Channel member sent me the attached model config.
[model10.zip](https://github.com/EdgeTX/edgetx/files/10135611/model10.zip)

Note line 253: srcRaw: S1
Note line 252: destCh: 8

Note line 237: destCh:10

Somehow, destCh:8 shows up AFTER destCh:10. Results in some strange behavior like the S1 knob changing output on Ch9. When I changed line 252 from destCh: 8 to destCh:11 (next unused mix line) the entry was parsed in the mix page correctly on the radio.

### Expected Behavior

Seems the destCh: value needs some error handling as this control did not appear ANYWHERE in the model configuration, yet it sent output and appeared in the mixer.

### Steps To Reproduce

Edit modelx.yml
Change destCh to be out of order. I do not know how this occurred for the user initially.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Looks like I picked a bad week to quit amphetamines.

raphaelcoeffic, 2022-12-11:
@jmxp69 this is basically the internal representation of the mixer lines. You'd better not mess with them ;-)

The underlying issue here is that this internal representation makes a couple of assumptions that should not be messed up with. The solution should be to avoid presenting these in the YAML format and instead load the mixer lines one after another without caring too much about that. That will require however some brain cells to be done properly and in a backward-compatible way.

As simple trick would be to check the order of mixer channels after loading and reordering in case there is something messed-up.

jmxp69, 2022-12-11:
I didn't change them. A viewer of my channel sent me his model file and has no idea how it happened. I simply found the issue with Ch:8 showing up after Ch:10 in the file. That is the root of the problem. My suggestion is that there should be some sort of sanity check to adjust the model configuration if this happens.

raphaelcoeffic, 2022-12-11:
> A viewer of my channel sent me his model file and has no idea how it happened

I could believe this is resulting from some earlier version where we had some errors with copy-paste or something like that. That being said, yes, *sorting after loading* is basically what your suggestion amounts to.

frank-hubgit, 2022-12-30:
Hi,
I think that the behavior at my x12s and tx16smk2 is similar.
I copied an existing model, reordered the channels and copied it to the x12s.
At mixes I can see the channels 1,4 and 5. Channels 2 and 3 are not shown but their functions are working.
companion:
[image]
x12s:
[image]
the yaml file 
[Falcon.yml.zip](https://github.com/EdgeTX/edgetx/files/10325376/Falcon.yml.zip)
Hopefully this belongs to this issue.

raphaelcoeffic, 2022-12-31:
@frank-hubgit how did you re-order the channels? Can you reproduce it?

frank-hubgit, 2022-12-31:
I did not use the perfect companion way  (copy an existing model, moving the channels at output and clearing the unused).
I copied the mixes manually from one channel to another one and renamed the channel at output afterwards.
Doing it the perfect (output-moving) way: copy an existing model, moving the channels at output window and then clearing and reconfiguring works correctly.
Doing it the other way: mixes, cut the channel and paste worked now too.
So I could not reproduce it. 
The yaml-files Original, Test-output and Test-mixes are beside.
[Original.yml.zip](https://github.com/EdgeTX/edgetx/files/10327170/Original.yml.zip)
[Test-mixes.yml.zip](https://github.com/EdgeTX/edgetx/files/10327171/Test-mixes.yml.zip)
[Test-output.yml.zip](https://github.com/EdgeTX/edgetx/files/10327172/Test-output.yml.zip)

raphaelcoeffic, 2023-02-10:
I think we should do the following:
- find the issue in Companion (thus having ordered channels in YAML; could be sufficient to order them before output YAML)
- re-order after loading in radio in case that happens, and to be able to recover from that situation (could be kind of tricky)

elecpower, 2023-02-10:
> I think we should do the following:
> 
> * find the issue in Companion (thus having ordered channels in YAML; could be sufficient to order them before output YAML)
> * re-order after loading in radio in case that happens, and to be able to recover from that situation (could be kind of tricky)
> 
> @elecpower are you on board? I'll take care of having the sort done while loading on radio, but it would be kind if you could take care of the sort on Companion.

I've got the Companion part.

commit d5857c0d (2023-02-10): fix: sort mixer lines after loading model

Fixes #2806
PR #3172 (2023-02-15): fix: sort mixer lines after loading model
Fixes #2806

~~Totally untested so far :-D~~ seems to work for me 👻

Btw. I know the worst case is $O(n^2)$, but it should not ever happen. The goal of this is to avoid spending too much FLASH on something hopefully useless.
commit 36959302 (2023-02-15): fix: Sort mixer lines after loading model (#3172)

* fix: sort mixer lines after loading model

Fixes #2806

* fix(cpn): sort mixer lines after loading model

* More efficient swap

---------
commit ab162184 (2023-02-15): fix: Sort mixer lines after loading model (#3172)

* fix: sort mixer lines after loading model

Fixes #2806

* fix(cpn): sort mixer lines after loading model

* More efficient swap

---------
closed 2023-02-15 by pfeerick