EdgeTX/edgetx #2810: Synchronization failed while creating the model in WIZARD.

al1965al, 2022-12-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Synchronization failed while creating the model -wizard - plane. 
Appears on Edge TX 2.7.1 and 2.8

https://www.youtube.com/watch?v=_PE6sHLRN58

### Expected Behavior

Appears on Edge TX 2.7.1 and 2.8
TX16S. Bought recently from Amazon at the Radiomaster store.
Also, the error is detected in many people who bought the transmitter 1.5 years ago.
Most likely a software error.

### Steps To Reproduce

https://www.youtube.com/watch?v=_PE6sHLRN58

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

al1965al, 2022-12-03:
In video easy to notice that the wizard has a very long response delay.

eshifri, 2022-12-03:
Can you please clarify what is wrong here?

al1965al, 2022-12-03:
https://www.youtube.com/watch?v=_PE6sHLRN58

eshifri, 2022-12-03:
Yes, I have watched it several times.
What is do mean by synchronization loss? Amd what is the expected behavior?

al1965al, 2022-12-03:
https://ibb.co/P4XykY4

Can't you see how the whole screen is filled with one color?
Can't you see the delay?

eshifri, 2022-12-03:
Now I understand what you mean.
Wizard is a LUA script. The delay is caused by the start of the script.
Garbled screen is also, most likely, a script "artifact" :-)
Thank you for bringing it up.

al1965al, 2022-12-03:
Many pilots have already written to me that this error occurs. On the TX16S of a different year of manufacture. This is not critical. BUT I want everything to work smoothly and correctly.
Thanks

pfeerick, 2022-12-04:
It is a cosmetic issue only - the first frame of the Lua double-display buffer isn't cleared before it's shown. So it will either have garbage info, or will show the last frame from the last standalone lua run.

Duplicate of #635

JimB40, 2022-12-05:
Confirmed. The more MCU intensive LUA is at init the more you can see it.
1. If you run LUA just after startup kind of rainbow garbage is shown
2. If you run after running previous LUA was running, kind of screenshot of this previous script is visible for short period.
So definetely display buffer is not cleared or initialized before runnin LUA in standalone mode.

PR #2916 (2022-12-28): fix(color): Clear Lua lcd buffer and display 'Loading...'

Fixes #2810

Summary of changes:

Clear the Lcd buffer and display "Loading..." text in the constructor for StandaloneLuaWindow
closed 2022-12-28 by pfeerick