EdgeTX/edgetx #2811: Celo75

Celo75, 2022-12-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The menu in the Czech language does not display letters written with diacritics. Letters with diacritics do not appear in all menu tabs. For example: Čeština is displayed as estina, or všechny is displayed as vechny.

### Expected Behavior

Displaying accented characters.

### Steps To Reproduce

1. Firmware in Czech language.
2. Viewing, for example, a telemetry card.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]
[image]

raphaelcoeffic, 2022-12-10:
@Celo75 seems like the Latin1-Extended-A charset was missing, which is mostly (but not exclusively) affecting the Czech translation.

pfeerick, 2022-12-11:
Should be fixed in the next nightly and 2.8.1 when released.

PR #2841 (2022-12-11): fix(cz): Add Latin1 Extended-A (0x100-0x17F)
Fixes #2811
commit d4174041 (2022-12-11): fix(cz): Add Latin1 Extended-A (0x100-0x17F) (#2841)

Fixes #2811
commit 042388ec (2022-12-13): fix(cz): Add Latin1 Extended-A (0x100-0x17F) (#2841)

Fixes #2811
closed 2022-12-11 by pfeerick