EdgeTX/edgetx #2816: Q X7S ACCST won't boot to bootloader with EdgeTX 2.8

mdr92, 2022-12-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The EdgeTX Bootloader won't load after flashing it with the 2.8 Bootloader, EdgeTX 2.7.1 Bootloader works as does the OpenTX Bootloader, but holding the trim buttons and powering on the radio it just boots the radio normally to the current firmware.

### Expected Behavior

The radio should boot to the bootloader when the trims are held in and the radio is powered on.

### Steps To Reproduce

Start up Radio
Write EdgeTX 2.8 Bootloader
Power off Radio
Hold in [horizontal] Trims
Power on radio
Radio boots up into EdgeTX 2.7.1 (or other previous firmware).

### Version

2.8.0

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Anything else?

I have tested this from OpenTX and EdgeTX 2.7.1, The EdgeTX 2.8 Bootloader flashes ok but then can't be booted to it, the radio starts up normally. As my Radio has a faulty USB Port this means I'm unable to update to EdgeTX 2.8 as I can't use that version bootloader.

pfeerick, 2022-12-04:
You should be able to use the 2.7.1 bootloader to flash 2.8.0, but there is obviously some issue going on here still if the bootloader isn't working properly :-/

mdr92, 2022-12-04:
> 

Unfortunately the 2.7.1 bootloader won't flash 2.8, it says "Not a valid firmware". I've made a bit of progress though, no matter how many times I flash the bootloader for 2.8 or try and flash the 2.8 firmware from the 2.7.1 bootloader it fails, that's using "x7-v2.8.0.bin" downloaded from the EdgeTX Buddy website. I just downloaded the full zip pack of all 2.8 firmwares instead and flashed the bootloader with "x7-f6d140e.bin" instead and then that booted fine and then flashed 2.8 Firmware fine.

The baffling thing is both have an identical MD5 of "b815b3d8387495e9f6b0cfefe9d4bf06" so I've no idea why one works and one doesn't?

gagarinlg, 2022-12-04:
The 2.7.1 bootloader does support flashing 2.8 without problems. Something must have gone wrong before.

Bebbetufs, 2023-04-10:
I'm having the same issue. I've tried the downloaded FW from Buddy as well as from the multi language zip. 
I also tried 2.7.1 as per the suggestions above. I've tried many different combinations of clicking, holding and tapping the on button. 
The radio still boots normally and the flashing of the bootloader seems to work correctly.

gagarinlg, 2023-05-05:
Buddy had a bug where it downloaded the wrong firmware and Buddy is now discontinued.
Can you please download the release via the EdgeTX github page and try again?

At least one user has no issues entering the bootloader.

closed 2023-05-26 by gagarinlg