EdgeTX/edgetx #2821: JP Firmware does not exist.

ToshihiroMakuuchi, 2022-12-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

@pfeerick

> EdgeTX "Flying Dutchman" v2.8.0 - language firmwares
> Currently, CZ, CN, TW and JP are not built for B&W as they either didn't display properly on B&W when last checked, or resulted in the firmware being too big.
> 

JP firmware does not exist, was there some underlying problem?
Is there anything I can do?

### Expected Behavior

CZ, CN, and TW exist, but only JP does not.
Is it not just a B&W issue?

### Steps To Reproduce

I unzipped many of the ZIP files and checked the contents, but no JP firmware was found.

### Version

2.8.0

### Transmitter

Jumper T16

pfeerick, 2022-12-06:
Pretty sure I just forgot to add it to the list - the build script I'm using over at https://github.com/pfeerick/lang-firmwares (which you can see there in the root directory) is a lighter form of the script I wrote some 18 months ago to do the firmware builds locally, and I didn't add `JA` to the list of builds for some reason. It's in now, so lets see how the build goes :)

https://github.com/pfeerick/lang-firmwares/actions/runs/3634366752

pfeerick, 2022-12-07:
Ok, that failed due to monkey behind keyboard... I did JA instead of JP...

However, looks like missed JP in #2454 and #2511 so need translations for the following before 2.8 will build for JP...
```
#define TR_WIDGET_GAUGE                "Gauge"
#define TR_WIDGET_MODELBMP             "Models"
#define TR_WIDGET_OUTPUTS              "Outputs"
#define TR_WIDGET_TEXT                 "Text"
#define TR_WIDGET_TIMER                "Timer"
#define TR_WIDGET_VALUE                "Value"
 
#define TR_WIDGET_FULLSCREEN           "Full screen"
```

ToshihiroMakuuchi, 2022-12-07:
@pfeerick 
For now, here is the Japanese translation.

JP Translation

#define TR_WIDGET_GAUGE                "ゲージ"
#define TR_WIDGET_MODELBMP        "モデル"
#define TR_WIDGET_OUTPUTS            "出力"
#define TR_WIDGET_TEXT                   "テキスト"
#define TR_WIDGET_TIMER                 "タイマー"
#define TR_WIDGET_VALUE                 "値"
 
#define TR_WIDGET_FULLSCREEN      "フルスクリーン"

pfeerick, 2023-02-19:
Closing as JP firmware builds are present in the 2.8.1 language firmware release :)

closed 2023-02-19 by pfeerick